fossi-foundation/open-pdks #2: Incorrect descriptions for sky_fd_pr_rf libraries

mithro, 2020-07-09:
https://github.com/RTimothyEdwards/open_pdks/blob/7ec769750bff68becb2305893821a7bbd8df8118/sky130/README#L175-L176

The following descriptions are wrong;
```
 sky130_fd_pr_rf/		RF primitive devices w/fixed layout 
 sky130_fd_pr_rf2/		Additional RF primitive devices w/fixed layout 
```

It should be something like the following;
```
 sky130_fd_pr_rf/		Older library of RF primitive devices w/fixed layout for compatibility (don't use in new designs).
 sky130_fd_pr_rf2/		Current library of RF primitive devices w/fixed layout (use in new designs).
```

RTimothyEdwards, 2020-07-09:
@mithro :  Last I saw from the interim skywater-random repository, pr_rf contained all the primitive devices for transistors, and pr_rf2 contained a few cells possibly overlapping with pr_rf plus a lot of cells of unknown origin and unknown usability.  For example, if you consider pr_rf to be somehow deprecated, then you will not have any bipolar transistors in the process.

RTimothyEdwards, 2020-10-20:
This should have all been resolved since the sky130_fd_pr repository was released, so I'm closing this now.

closed 2020-10-20 by RTimothyEdwards