fossi-foundation/open-pdks #4: KLayout layer properties missing layer 83/44 and 236/0

smunaut, 2020-07-09:
I'm just loading some examples from the standard cell libraries ( the `hd` variant ) and in those you find layer `83/44` which is some pure text layer.  And then layer `236/0` which might be equivalent to `235/4` (which is not used in the std cell library AFAICT) ?

RTimothyEdwards, 2020-07-09:
I don't use klayout myself (too busy hacking on Magic) to be too familiar with the format.  I'm relying on klayout setup files given to me by others.  Layer 236:0 is not parsed by either the DRC or LVS Calibre decks, so it is unknown/unknowable.  Is it used like a boundary?  I.e., is there good evidence to suggest that it might be intended to be some equivalent to prBndry?

smunaut, 2020-07-09:
In all cells I've looked at until now, the 236:0 has the same content as 81:4.  prBndry doesn't seem to appear anywhere.

snx90, 2020-08-26:
Hi Timothy,

I’ve been using KLayout as well. When I tried to open the GDS with Magic I was unable to visualize the result. What are the steps needed?