fossi-foundation/open-pdks #5: Command "make" fails with Error:  "No layout file in either mag/ or maglef/"

rbarzic, 2020-07-16:
Hi,

I have tried to follow the instructions for generated a sky130 pdk.
I've checkouted skywater-pdk (commit 4e5e318e0cc578090e1ae7d6f2cb1ec99f363120)
and run the make timing command

Then, in open_pdks/sky130 (commit 282d954a5bd4714241320b6e49d1108a72fbd81d), I have run the  "make" command - I have EF_STYLE=0 in the Makefile)
magic version is 8.3.33

The  make command ends with the following message: 
# Add correct bounding boxes on Magic layouts
../common/insert_property.py -std_format  `pwd`/sky130A sky130_fd_io sky130_fd_io_top_gpio_ovtv2 \
	"FIXED_BBOX 0 407 28000 40000"
Error:  No layout file in either mag/ or maglef/
(/home/roba/perso/github/open_pdks/sky130/sky130A/libs.ref/sky130_fd_io/mag[lef]/sky130_fd_io_top_gpio_ovtv2.mag)
../common/insert_property.py -std_format  `pwd`/sky130A sky130_fd_io sky130_fd_io_top_xres4v2 \
	"FIXED_BBOX 0 407 15000 40000"
Error:  No layout file in either mag/ or maglef/
(/home/roba/perso/github/open_pdks/sky130/sky130A/libs.ref/sky130_fd_io/mag[lef]/sky130_fd_io_top_xres4v2.mag)
make[1]: Leaving directory '/home/roba/perso/github/open_pdks/sky130'
echo "Ended sky130A PDK staging on "`date` >> sky130A_install.log

(note that the error message is not present in the log file)

When going through the log, the first errir I noticed is : 

Writing 'sky130_fd_pr_rf_xcmvpp2_nwell'
Writing 'balun'
Writing '(UNNAMED)'
Must specify name for cell (UNNAMED).
Done.
Error message output from magic:
Can't write file named '(UNNAMED)'
No LEF file install;  need to generate LEF.
Removing existing library /home/roba/perso/github/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr_base/lef/sky130_fd_pr_base.lef
Searching for supported devices in PDK script /home/roba/perso/github/open_pdks/sky130/sky130A/libs.tech/magic/sky130A.tcl.
PDK library is sky130
Getting GDS file list from /home/roba/perso/github/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr_rf/gds.
Creating magic generation script to generate magic database files.
No source for abstract views:  Abstract views not made.
Running magic to create magic database files.

Do you have some commit numbers for open_pdks, skywater-pdk and magic that are verified to work together for you so I could try again on my side ?

RTimothyEdwards, 2020-07-17:
This is "correct" behavior for now.  There are libraries missing from the Google/SkyWater repository that should go up within the next month.  I have not bothered to edit the Makefile to avoid the problem.  The open_pdks should be working correctly to install the five standard cell libraries that are currently in the repository.  The errors will go away (or will be fixed) when the missing libraries (sky130_fd_pr_base, sky130_fd_pr_rf, sky130_fd_pr_rf2, and sky130_fd_io) are added to the repository.

rbarzic, 2020-07-17:
ok. But then I got this error when running make install-dist (or make install-local) :

```
cho "Starting SKY130 PDK migration on "`date` > sky130A_migrate.log
../common/staging_install.py -std_format -source `pwd`/sky130A \
	-target /home/roba/perso/github/nanorv32_sky130/sky130A \
	-local /home/roba/perso/github/nanorv32_sky130_local/sky130A \
	-link_from none |& tee -a sky130A_migrate.log
Installing in target directory /home/roba/perso/github/nanorv32_sky130/sky130A
Removing files from target
Copying staging files to target
Done.
Changing local path references from /home/roba/perso/github/open_pdks/sky130/sky130A to /home/roba/perso/github/nanorv32_sky130_local/sky130A
Part 1:  Tools
      magic (1 substitution)
      qflow (8 substitutions)
Traceback (most recent call last):
  File "../common/staging_install.py", line 419, in <module>
    print('Replacing files with symbolic links to ' + link_from + ' where possible.')
TypeError: must be str, not NoneType
echo "Ended SKY130 PDK migration on "`date` >> sky130A_migrate.log

```
Is this related to the missing libraries you mentionned above ? If yes, does that means we can't use open_pdks right now ?  (for those who son't have access yet to the missing libraries )

rbarzic, 2020-07-17:
After a bit of debugging, look like LINK_TARGETS=none is not supported by the code 

staging_install.py around line 273
```pyhton
  link_name = None
    for option in optionlist[:]:
        if option[0] == 'link_from':
            optionlist.remove(option)
            if option[1].lower() == 'none':
               link_from = None
```
then 

```python
   if link_from != 'source':
        thispdk = os.path.split(targetdir)[1]
        print(f"-D- link_from = {link_from}")
        # Only create links for PDKs other than the one we are making links to.
        if thispdk != link_from:
            print('Replacing files with symbolic links to ' + link_from + ' where possible.')
            for techdir in techdirs:
```

If I use "sourse" instead of none, the make install-dist seems to be executed correctly but the DIST_PATH directory seems to be incomplete (no lib/lef files installed for example)
comman used:
```
make install-dist  EF_STYLE=0 LINK_TARGETS=source STAGING_PATH=/home/roba/tmp/sdk130_staging SKYWATER_PATH=/home/roba/perso/github/skywater-pdk LOCAL_PATH=/home/roba/perso/github/nanorv32_sky130_local DIST_PATH=/home/roba/perso/github/nanorv32_sky130
```

RTimothyEdwards, 2020-07-20:
Sorry for taking so long about it, but I have committed a fix for this error.  Thanks for reporting it in detail.  I have patched the source on the opencircuitdesign.com git repo master branch, and the github repo will be automatically mirrored overnight.  I will close this issue after I have had a chance to check and confirm the fix.

RTimothyEdwards, 2020-08-02:
This should be okay now.  Also, the missing libraries are now commented out of the Makefile and will remain that way until they are in the Google/SkyWater repository, at which time I will uncomment them.  This will prevent the error messages from popping up.  I am closing the issue now.

closed 2020-08-02 by RTimothyEdwards