fossi-foundation/open-pdks #7: Missing scripts?

mithro, 2020-07-27:
@RTimothyEdwards you said on slack;

> The latest push of open_pdks (github.com/RTimothyEdwards/open_pdks) has individual python scripts to build libraries for each of the file formats.  These are all self-contained.  Namely, the ones you want are "create_lef_library.py", "create_spice_library.py", and "create_gds_library.py".  Do not use "create_lib_library.py" (unfinished and probably not useful) or "create_verilog_library.py" (will not work with all the complicated sets of files you have, and the definitions and include statements)

But I can't find them in this repo?

RTimothyEdwards, 2020-07-27:
@mithro :  The automatic mirror process runs at 3:00am EDT.  If you are looking for them prior to that, pull them from opencircuitdesign.com.

RTimothyEdwards, 2020-08-02:
All scripts should be present now.  Closing this issue.

closed 2020-08-02 by RTimothyEdwards