fossi-foundation/open-pdks #20: LIB_MIN / LIB_MAX swapped in cells configs

smunaut, 2020-07-30:
`open_pdks/sky130/openlane/sky130_fd_sc_hd/config.tcl` contains : 

```
set ::env(LIB_MAX) "$::env(PDK_ROOT)/TECHNAME/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib"
set ::env(LIB_MIN) "$::env(PDK_ROOT)/TECHNAME/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ss_100C_1v60.lib"
```

But that's not good. `LIB_MAX` is used for maximum delay computations ( for setup time analysis ) and so this should be the slower possible corner ( `SS` at high temp with low voltage )
And similarely, `LIB_MIN` is used for minimum delay computations ( for hold time analysis ) and so this should be the fastest possible corner ( `FF` at low temps with high voltage )

riking, 2020-07-31:
Suggestion from Slack: rename to SLOWEST / FASTEST.

smunaut, 2020-07-31:
OpenSTA also use "Best Case" and "Worst Case" 
```
read_liberty -corner bc XXX
read_liberty -corner typ XXX
read_liberty -corner wc XXX
```

RTimothyEdwards, 2020-08-02:
The config.tcl files now use LIB_FASTEST and LIB_SLOWEST, so I am closing this issue.

closed 2020-08-02 by RTimothyEdwards