fossi-foundation/open-pdks #24: Make process stuck at foundry_install.py

snx90, 2020-08-21:
The make process (ubuntu machine) is stuck in the following step:

> # Install base device models from vendor files
> # (NOTE: .mod and .pm3 files should not be in /cells/?)
> set -f ; ../common/foundry_install.py -std_format -source /home/santos/Documents/PDKs/skywater-pdk/libraries -target `pwd`/sky130A \
> 	-ngspice sky130_fd_pr_base/v%v/models/* filter=custom/scripts/fixspice.py \
> 	-ngspice sky130_fd_pr_base/v%v/cells/*.mod filter=custom/scripts/fixspice.py \
> 	-ngspice sky130_fd_pr_base/v%v/cells/*.pm3 filter=custom/scripts/fixspice.py \
> 	|& tee -a sky130A_install.log

snx90, 2020-08-26:
Hi Timothy,

The problem was with magic. Apparently, the installation was not fully successful in my machine so it was returning an error when invoked with the '--version' flag. Since this error case is not considered in the 'foundry_install.py' script, the make process was getting stuck at that point.

RTimothyEdwards, 2020-08-31:
What does it return, so that I can catch that in the foundry install script?

By the way, Tim Ansell and I are still plugging away at the primitive devices, but we have made excellent progress over the last week (involving several thousand lines of python code), and hopefully we can complete that work this week.

RTimothyEdwards, 2020-10-20:
This was resolved a while ago, and although I left the issue open while I was still developing the install script around the latest additions to skywater-pdk, it should all be better now, so I will go ahead and close this issue.

closed 2020-10-20 by RTimothyEdwards