fossi-foundation/open-pdks #26: QFlow Fails Due to ABC String Length Assertion

yrrapt, 2020-08-26:
Running a design through QFlow fails during synthesis due to ABC stage due to the following assertion fail:

```ABC: Scl_Libertyosys-abc: src/map/scl/sclLiberty.c:344: Scl_LibertyReadString: Assertion `Pair.End-Pair.Beg < ABC_MAX_LIB_STR_LEN' failed.```

I worked around this by just increasing the `ABC_MAX_LIB_STR_LEN` in the ABC source files and recompiled Yosys but I suspect there's a more elegant way to get around this issue.

RTimothyEdwards, 2020-08-31:
This is not specifically a qflow issue;  you would need to take this issue up with Claire Wolf (yosys) or Alan Mischenko (abc).  I've never seen this issue before. . .  Just how long are the strings that are causing this?

RTimothyEdwards, 2020-10-20:
Given that this issue is not open_pdks-specific, and not even qflow-specific, I am going to close this issue now.  If you have any objections to that, feel free to re-open it.

closed 2020-10-20 by RTimothyEdwards