fossi-foundation/open-pdks #27: sky130 layer thickness and elevation information

Piyush3dB, 2020-09-02:
The [process stack diagram](https://skywater-pdk.readthedocs.io/en/latest/rules/assumptions.html#process-stack-diagram) shows the layer thickness and elevation.

Is it possible to obtain this information for all possible layers in a textual format?  Or is it possible to compute these from existing data?

Any pointers will be useful. Thanks!

Piyush3dB, 2020-09-02:
For the record I found the info that @RTimothyEdwards is referring to here onwards:
[https://github.com/RTimothyEdwards/open_pdks/blob/52f78fa08f91503e0cff238979db4589e6187fdf/sky130/magic/sky130.tech#L1257](https://github.com/RTimothyEdwards/open_pdks/blob/52f78fa08f91503e0cff238979db4589e6187fdf/sky130/magic/sky130.tech#L1257)

Might find extra info elsewhere so will update in due course.  Cheers!

Piyush3dB, 2020-09-21:
@RTimothyEdwards 
Does anyone know an efficient way to map the GDS height records in 

https://github.com/RTimothyEdwards/open_pdks/blob/52f78fa08f91503e0cff238979db4589e6187fdf/sky130/magic/sky130.tech#L5511

to the layer information in the CSV file at
[https://skywater-pdk.readthedocs.io/en/latest/_downloads/8732cefe49e90050bd82dce54d3aa8dc/gds_layers.csv](https://skywater-pdk.readthedocs.io/en/latest/_downloads/8732cefe49e90050bd82dce54d3aa8dc/gds_layers.csv)

Take for instance height information from `sky130.tech`:
`
  "mimcap": {
    "name": "mimcap",
    "position": 2.4661,
    "height": 0.2
  },
`

Which `Layer:Datatype` does this map to?  

The `gds_layers.csv` CSV file has the following entries which closest match 'mimcap:

[image]

By inspection, I can infer which heights map to which `Layer:Datatype` combo (and there might be multiple it seems), but isn't this information already present somewhere?  

If not, what is the algorithm to reconcile the two sources of information?

Best Regards

RTimothyEdwards, 2020-09-21:
@Piyush3dB :  That would be a bit difficult because the magic techfile is created separately by hand rather than from the SkyWater repository data.  This is a bit of a chicken-and-egg problem because Magic is used to create some of the SkyWater repository.

The actual relationship is this:  Layer "mimcap" in the "height" record appears in the "cifoutput" section as

   layer CAPM     *mimcap
        labels  mimcap
        calma   89 44

Indicating that it maps to GDS layer:purpose pair 89:44.  There is not necessarily a 1-to-1 relationship between a magic layer and a GDS layer, although there is in this case.

RTimothyEdwards, 2020-10-20:
Is this resolved to the point where I can close the issue?

Piyush3dB, 2020-10-24:
Matter resolved. Thanks for your help!

closed 2020-10-24 by Piyush3dB