fossi-foundation/open-pdks #28: sky130_fd_sc_lp is missing setup files for openlane

snx90, 2020-09-07:
PDK_ROOT/sky130A/libs.tech/openlane does not have the corresponding sub folder with the openlane setup files of sky130_fd_sc_lp

RTimothyEdwards, 2020-10-20:
@snx90 :  You would need to take that up with @ax3ghazy .  However, I believe that it is still true that the LP and HVL libraries are still not officially submodules of skywater-pdk, and so still not officially supported, although we're pushing to get that done.

mguthaus, 2023-05-19:
This issue is still outstanding. Any update from @ax3ghazy ?

RTimothyEdwards, 2023-05-19:
Ahmed left the company some time ago.  There _is_ continuing support for the other variants, but someone may have dropped the ball on the LP library.  Need to check with @marwaneltoukhy or Mohamed Shalan.

marwaneltoukhy, 2023-05-21:
@kareefardi can you please take a look?