fossi-foundation/open-pdks #43: No setup commands in `README.md` file

mithro, 2020-10-25:
There are not any quick start instructions on how to get started with the `open_pdks`. It needs something like;

```shell
git clone https://github.com/RTimothyEdwards/open_pdks.git
cd open_pdks
./configure --with-sky130-source=<directory where you cloned https://github.com/google/skywater-pdk repository>
cd sky130
make tools-a
```

Unclear to me if the above is correct. 

It is unclear to me if anything needs to be done in the `google/skywater-pdk` repository before using `./configure` -- aka do you have to run `make timing` or?

RTimothyEdwards, 2020-10-25:
It says "Instructions" and points you to the README in the "target foundry process" which would in this case be sky130.
So if you follow that and go look at sky130/README, you'll see the full set of installation instructions.  Except that those are not the instructions that you, specifically, want to follow (see below).

The scattershot location of instructions are largely a result of several overhauls of the code since its inception.  Cleanup will happen eventually.

The above is sort of correct;  it's just that what you want out of it is the magic techfile, which is why I directed you to cd to sky130 and do "make tools-a".  Normally one just does "make" from the top level, but since that presupposes that the skywater-pdk repo is installed, then it's not what you want to do to get just the techfile for using to build the skywater-pdk repo.  The instructions you want to follow are atypical.  Likewise, there is no need for you to have the skywater-pdk or run "make timing".  You don't even need to have "--with-sky130-source=" pointing to anything relevant;  it just needs to be set to something non-null to keep the make script from failing.

RTimothyEdwards, 2020-10-25:
Fixed this, too.  It will show up in github in a few hours.

mithro, 2020-11-03:
I've yet to be able to make the instructions work....

RTimothyEdwards, 2021-08-22:
This has been reworked several times, most recently in #127, so I'm closing this issue.

closed 2021-08-22 by RTimothyEdwards