fossi-foundation/open-pdks #45: Build breaks when not building the sky130_osu_sc_t18 library

agorararmard, 2020-10-29:
The build breaks when the sky130_osu_sc_t18 library is not there, because it tries to move the tlef file with the mv command without checking if the file exists or not, and so the build fails if it's not there (which will always be the case until the library is released).
```
mv `pwd`/sky130A/libs.ref/sky130_osu_sc_t18/techlef/sky130_osu_sc.tlef `pwd`/sky130A/libs.ref/sky130_osu_sc_t18/techlef/sky130_osu_sc_t18.tlef

mv: cannot stat `/home/travis/build/RTimothyEdwards/open_pdks/sky130/sky130A/libs.ref/sky130_osu_sc_t18/techlef/sky130_osu_sc.tlef': No such file or directory
```

The solution to this issue is in this PR #39, but it was closed without being merged :-)

RTimothyEdwards, 2020-10-29:
I merged that, so I don't know why it appears either not to have happened, or reverted back to the way it was previously.  Anyway, I just did another commit to the master repo on opencircuitdesign.com with an attempt to do the renaming inside the foundry_install.py script.  Please check if it is correctly preserving the .tlef extension or not.

agorararmard, 2020-10-29:
Roger that.

agorararmard, 2020-10-29:
@RTimothyEdwards: an extra case was needed in the nested if handling the fileext. I added that in #46 .

RTimothyEdwards, 2020-11-02:
Should be fixed now (will not appear in github until tomorrow morning, though).  I will leave the issue open until it is known good in the github repo.

agorararmard, 2020-11-03:
Looks good to me!

closed 2020-11-03 by agorararmard