fossi-foundation/open-pdks #48: sky130.tech: rmp devices not extracted in default style

d-m-bailey, 2020-11-03:
In the sky130/magic/sky130.tech file, rmp devices are extracted in the ngspice(orig) style, but not the default ngspice() style.

Is this intentional?

For example, with ngspice(orig), we get 
.subckt sky130_fd_sc_hd__conb_1 VGND VNB VPB VPWR HI LO
R0 HI VPWR sky130_fd_pr__res_generic_po w=480000u l=45000u
R1 VGND LO sky130_fd_pr__res_generic_po w=480000u l=45000u
.ends

But with the default ngspice(), we get nothing - just the extracted instance.
X_31159_ VGND VGND VPWR VPWR _31159_/HI mem_la_addr[0] sky130_fd_sc_hd__conb_1

RTimothyEdwards, 2020-11-03:
Thanks for pointing this out. . . I have some other minor fixes to make to the tech file and will update shortly.

d-m-bailey, 2021-12-27:
There have been no further issues with this in the past year, so I'm closing this issue.

closed 2021-12-27 by d-m-bailey