fossi-foundation/open-pdks #49: open_pdks not able to find my skywater-pdk repository

mithro, 2020-11-03:
```
$ ./configure --with-sky130-source=$HOME/github/google/skywater-pdk/
configure: Found technology directories: sky130
configure: Checking whether source path is specified for 'sky130'
configure: Checking specified path for 'sky130' at /home/tim/github/google/skywater-pdk
checking for /home/tim/github/google/skywater-pdk... yes
configure: 'sky130' source path found at /home/tim/github/google/skywater-pdk
configure: Checking whether local path is specified for 'sky130'
configure: error: Option --with-sky130-local-path=<path> not specified!
```

c2662b9991101dc64f8d2fd850bb5996ad5d1f15

RTimothyEdwards, 2020-11-04:
So give it the --with-sky130-local-path=<path> option, as it says.

If you want to use the staging area as the install location, then this can be a throwaway value;  it is only meaningful if you do "make install".

mithro, 2020-11-04:
It sounds like `--with-sky130-local-path` is equivalent to the normal `--prefix` parameter you would give to normal tools?

Is there any reason not to have a default value for the `--with-sky130-local-path` flag? IE `--prefix` normally defaults to `/usr/local/`. Is there a reasonable default that various tools would find things at by default?

Otherwise, maybe if you don't provide `--with-sky130-local-path` and it just disable the `make install` target?

RTimothyEdwards, 2020-11-04:
There is probably a long and worthwhile discussion to be had here.  There is no consensus on where EDA tools get their PDK data from.  The efabless /ef/tech/ path is its own thing which I would not force on anyone.  Long ago, magic used to assume /home/cad/ and I sort of like the idea of /usr/local/cad or /usr/share/cad or something like that.  I get rather annoyed by a lot of EDA tools that want to install themselves into my home directory and then install all project data mixed into the same space---ugh---so I want to avoid that.  I think --prefix=/usr/local by default followed by maybe "cad/" or "pdk/" and then "sky130A/" would work.  There needs to be some distinction between the base directory for all (potential) PDKs vs. the base directory for the SkyWater PDK.

Disabling the "make install" target when "local-path" is not selected is also a good idea;  there is some (probably minor technical) reason that I didn't do that (i.e., didn't spend enough time on it to get it to work).

RTimothyEdwards, 2021-08-22:
Since #127 changed the Make process to use the standard --prefix, this should no longer be an issue.

closed 2021-08-22 by RTimothyEdwards