fossi-foundation/open-pdks #52: Give mithro Triage access to this repo

mithro, 2020-11-04:
Can haz pls?

RTimothyEdwards, 2020-11-05:
If I give you access to the github version of the repository, you have to show me how to pull those changes into opencircuitdesign.com without breaking my local system.  The best way to get fast updates it to send a pull request, which I will expedite, especially if you send me a slack message at the same time.

RTimothyEdwards, 2020-11-05:
I can do pull requests from my local computer and that works fine, so I assume I can do something similar to pull any changes made directly to the repository here.

mithro, 2020-11-05:
Triage just gives me access to issues, not the code.

RTimothyEdwards, 2020-11-05:
"Triage" is some official github thing, then?  How do I give you triage access?

mithro, 2020-11-05:
`Settings > Manage Access` you can then add people and set their access level.

mithro, 2020-11-05:
I can now change github issues :-).

closed 2020-11-05 by mithro