fossi-foundation/open-pdks #54: mag views have conflicting port numbers

d-m-bailey, 2020-11-07:
After the maglef generation step, the `open_pdks/common/foundry_install.py `file has
```
        # Now list all the .mag files generated, and for each, read the
        # corresponding file from the mag/ directory, pull the GDS file
        # properties, and add those properties to the maglef view. Also
        # read the CDL (or SPICE) netlist, read the ports, and rewrite
        # the port order in the mag and maglef file accordingly.

```
It looks like the port order is being rewritten here.
From clkbuf_16 mag file before annotation.
```
flabel pwell 29 -17 63 17 0 FreeSans 200 0 0 0 VNB
port 3 nsew ground input
flabel pwell s 46 0 46 0 0 FreeSans 200 0 0 0 VNB
port 3 nsew default input
flabel nbase 29 527 63 561 0 FreeSans 200 0 0 0 VPB
port 4 nsew power input
flabel nbase s 46 544 46 544 0 FreeSans 200 0 0 0 VPB
port 4 nsew default input
flabel metal1 29 -17 63 17 0 FreeSans 200 0 0 0 VGND
port 2 nsew ground input abutment
flabel metal1 29 527 63 561 0 FreeSans 200 0 0 0 VPWR
port 5 nsew power input abutment

```
Notice how some `flabel` lines have 13 fields while others (the ones with 's' as the third field) have 14.
After annotation:
```
flabel pwell 29 -17 63 17 0 FreeSans 200 0 0 0 VNB
port 3 nsew ground input
flabel pwell s 46 0 46 0 0 FreeSans 200 0 0 0 VNB
port 3 nsew default input
flabel nbase 29 527 63 561 0 FreeSans 200 0 0 0 VPB
port 3 nsew power input
flabel nbase s 46 544 46 544 0 FreeSans 200 0 0 0 VPB
port 4 nsew default input
flabel metal1 29 -17 63 17 0 FreeSans 200 0 0 0 VGND
port 4 nsew ground input abutment
flabel metal1 29 527 63 561 0 FreeSans 200 0 0 0 VPWR
port 4 nsew power input abutment

```
Looks the port number after the longer lines (the ones with the 's') are propagated to the following shorted lines regardless of the signal name.
From `open_pdks/common/foundry_install.py`
`flabrex = re.compile('flabel[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+[^ \t]+[ \t]+([^ \t]+)')`
`flabrex` only matches the longer lines.
So for the unmatched shorter `flabel` lines, the last port number for a long line is used. If there hasn't been a long line encountered, the port number stays the same.
The following works for long and short lines.
`flabrex = re.compile('flabel[ \t]+.*[ \t]+([^ \t]+)[ \t]*')`

RTimothyEdwards, 2020-11-07:
This is a magic issue, not open_pdks, and it was resolved (sort of) yesterday.  I took steps in the open_pdks scripts to work around the problem, but I still need to track down the memory corruption issue in magic that caused it in the first place.  Your detailed analysis above will help, thanks.  The "longer line" format shows another minor issue in magic, which is that the annotation of labels from a LEF file is resetting the flags like the "sticky" flag.  Although the open_pdks issue has been effectively resolved, I will leave this issue open until I have resolved both the known issues in magic.

RTimothyEdwards, 2020-11-10:
If I were not so frazzled, perhaps I would do the smart thing and read your posts carefully.  Testing your proposed fix now.

d-m-bailey, 2020-11-10:
@RTimothyEdwards Sorry, I need to practice more brevity. Anything I can do to help?

d-m-bailey, 2020-11-13:
@RTimothyEdwards looks good, you can close this if you want.

Incidentally, I put the delete UNNAMED cell statements back in and the only difference appears to be less warnings.

d-m-bailey, 2021-12-27:
There have been no further problems with this in the last year, so I'm closing this issue.

closed 2021-12-27 by d-m-bailey