fossi-foundation/open-pdks #56: Include git-describe --long information in all generated files

mithro, 2020-11-15:
It would be good if when you generated files like the `sky130A.tech` file, you included the current git revision. The best way to do this is with `git-describe --long`.

mithro, 2020-11-15:
You should also probably include `--dirty` and `--broken` too.

RTimothyEdwards, 2020-11-16:
@mithro : The last thing I did along these lines was to add annotated tags. . . where in my process should I do this?  Before/after setting the tag?

RTimothyEdwards, 2020-11-16:
@mithro :  Oh, I see what you mean.  I currently set "version" in the techfile to the datestamp (e.g., "20200508") but it could just as easily be set to the result of "git describe --long".

When you say "files like the sky130A.tech file", do you have other files in mind?

mithro, 2020-11-16:
Any file which this repo creates.

RTimothyEdwards, 2020-11-16:
@mithro :  Okay, done (for sky130A.tech).

RTimothyEdwards, 2021-08-22:
Closing this issue, which was implemented a while ago (but more recently, took this a step further by keeping an information file in the installation with the commits of every repository used by open_pdks to compile the final installed PDK).

closed 2021-08-22 by RTimothyEdwards