fossi-foundation/open-pdks #58: Some simulation models not built

mballance, 2020-11-17:
Hello,
    I initially filed an issue against openlane for this (https://github.com/efabless/openlane/issues/91), and @agorararmard suggested an issue should be filed here as well.

    I'm building open_pdk using the rc4 release of openlane. Here are the relevant commits for Skywater and OpenPDK:
SKYWATER_COMMIT: 5cd70ed19fee8ea37c4e8dbd5c5c3eaa9886dd23
OPEN_PDKS_COMMIT: 48db3e1a428ae16f5d4c86e0b7679656cf8afe3d
    The build basically appears to work, and I can successfully run the 'test' step from openlane. However, quite a few of the simulation models used by efabless/caravel are not present.

    I see the following error when I  build pdk_root:
```
Writing all magic database files^M
Writing 'sky130_fd_pr__cap_vpp_08p6x07p8_m1m2_lishield'^M
Writing 'sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5'^M
Writing 'sky130_fd_pr__cap_vpp_11p5x11p7_m1m2_noshield'^M
Writing 'sky130_fd_pr__rf_test_coil1'^M
Writing 'sky130_fd_pr__rf_test_coil2'^M
Writing 'sky130_fd_pr__rf_npn_05v5_W1p00L2p00'^M
Writing 'sky130_fd_pr__rf_test_coil3'^M
Writing 'sky130_fd_pr__cap_vpp_04p4x04p6_m1m2_lishield'^M
Writing 'sky130_fd_pr__pnp_05v5_W3p40L3p40'^M
Writing 'sky130_fd_pr__rf_npn_05v5_W1p00L1p00'^M
Writing '(UNNAMED)'^M
Must specify name for cell (UNNAMED).^M
Done.^M
Error message output from magic:^M
File sky130_fd_pr__rf_npn_05v5_W1p00L1p00.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_npn_05v5_W1p00L2p00.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__pnp_05v5_W3p40L3p40.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_test_coil1.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_test_coil2.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_test_coil3.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_11p5x11p7_m1m2_noshield.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_08p6x07p8_m1m2_lishield.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_04p4x04p6_m1m2_lishield.mag couldn't be read^M
No such file or directory^M
Can't write file named '(UNNAMED)'^M
Migrating LEF files to layout.^M
Warning:  No LEF files found in /project/fun/openlane/pdk_root/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr/lef^M

```
I did a quick check on a few of the files reported missing, and they do exist. I've attached the full log for completeness. Any help is much appreciated!

Best Regards,
Matthew

[log.docker.txt](https://github.com/RTimothyEdwards/open_pdks/files/5550809/log.docker.txt)

agorararmard, 2020-11-17:
@mballance: I'll copy the first part of my response here as well for Tim's convenience.

> So, here you're using a version of skywater pdk that doesn't have the IOs, with a version of open_pdks that cannot compile them properly (even if they were there).

You should be using the HEAD of open_pdks and the HEAD of skywater-pdk.

mballance, 2020-11-17:
Appreciate the help, @agorararmard. I'm closing this issue.

Best Regards,
Matthew

closed 2020-11-17 by mballance