fossi-foundation/open-pdks #59: generic resistors don't match in LVS

d-m-bailey, 2020-11-18:
The `open_pdks/sky130/magic/sky130.tech` file extracts generic resistors as R-devices.

Generic resistors
```
sky130_fd_pr__res_generic_l1
sky130_fd_pr__res_generic_m1
sky130_fd_pr__res_generic_m2
sky130_fd_pr__res_generic_m3
sky130_fd_pr__res_generic_m4
sky130_fd_pr__res_generic_m5
sky130_fd_pr__res_generic_nd
sky130_fd_pr__res_generic_nd__hv
sky130_fd_pr__res_generic_pd
sky130_fd_pr__res_generic_pd__hv
sky130_fd_pr__res_generic_po
```
Sample extraction (2 terminal resistor):
```
.subckt sky130_fd_sc_hd__conb_1 VGND VNB VPB VPWR HI LO
R0 HI VPWR sky130_fd_pr__res_generic_po w=480000u l=45000u
R1 VGND LO sky130_fd_pr__res_generic_po w=480000u l=45000u
.ends
```
The spice library has the 3 terminal X-resistor definitions.
```
.subckt sky130_fd_sc_hd__conb_1 VGND VNB VPB VPWR HI LO
X0 VGND LO VNB short w=480000u l=45000u
X1 HI VPWR VNB short w=480000u l=45000u
.ends
```
Maybe change the extraction to `rsubcircuits` and change the spice library to 2 terminal `sky130_fd_pr__res_generic_po` devices.

Also the variant(orig) sytle has a spelling error.
```
 device rsubcircuit sky130_fd_pr_res_generic_nd     ndiffres \
        *ndiff pwell,space/w  error l=l w=w
 device rsubcircuit sky130_fd_pr_res_generic_pd     pdiffres \
        *pdiff nwell    error l=l w=w
```

Probably should be `sky130_fd_pr__res_generic_nd` and `sky130_fd_pr__res_generic_pd` (2 '_' between the 'pr' and 'res')

RTimothyEdwards, 2020-11-19:
The spelling errors have been noted and I am fixing them now.

The resistor types takes some working through, because the models are in different places and some really are R type devices.  The changing PDK was a moving target for a while.  The "short" resistor is different from the others, and I have been adding that manually to the SPICE library because it is not part of the PDK.  The "rsubcircuit" device can take a substrate node, so it can generate 3-terminal devices in the netlist.

d-m-bailey, 2021-08-11:
@RTimothyEdwards Mostly fixed. The open_pdks scripts convert X devices to R devices, but leaves them with 3 terminals. I believe they should be 2 terminal devices with an option third terminal specified as `$SUB=net`.

Here is the current status of the spice library.
```
find /openLANE_flow/pdks/sky130A/libs.ref/. -name '*.spice' -exec grep -A4 -i 'subckt.*conb' {} /dev/null \; | sed 's,.*/,,'
sky130_fd_sc_hvl.spice:.subckt sky130_fd_sc_hvl__conb_1 VGND VNB VPB VPWR HI LO
sky130_fd_sc_hvl.spice-R0 HI VPWR VNB sky130_fd_pr__res_generic_po w=510000u l=45000u
sky130_fd_sc_hvl.spice-R1 VGND LO VNB sky130_fd_pr__res_generic_po w=510000u l=45000u
sky130_fd_sc_hvl.spice-.ends
sky130_fd_sc_hvl.spice-
sky130_fd_sc_hd.spice:.subckt sky130_fd_sc_hd__conb_1 VGND VNB VPB VPWR HI LO
sky130_fd_sc_hd.spice-R0 VGND LO VNB sky130_fd_pr__res_generic_po w=480000u l=45000u
sky130_fd_sc_hd.spice-R1 HI VPWR VNB sky130_fd_pr__res_generic_po w=480000u l=45000u
sky130_fd_sc_hd.spice-.ends
sky130_fd_sc_hd.spice-
```

d-m-bailey, 2021-12-27:
There have been no issues with this in the last couple months, so I'll close this issue.

closed 2021-12-27 by d-m-bailey