fossi-foundation/open-pdks #75: 45 Degree Metal DRC Errors

yrrapt, 2020-12-05:
Drawing 45 degree paths on metal 1 and 2 results in DRC errors but the DRC error states that 45 degree angles are allowed. Therefore I believe the DRC tool is not correctly interpreting the angles.
[image]

[45degree_test.gds.zip](https://github.com/RTimothyEdwards/open_pdks/files/5646804/45degree_test.gds.zip)

Simple test case attached.

RTimothyEdwards, 2020-12-05:
@yrrapt : Can you confirm that the angles are really 45 degrees?  A tiny difference in width vs. length will raise an error (both in magic and in calibre).  I have  had this problem myself.  How did you generate the angled wires?  It's possible that magic's "wire" command might have round-off issues;  if so, I'd need to fix them, but if you only have a few cases then you might be able to fix it by hand with "splitpaint" and "spliterase" commands.

yrrapt, 2020-12-05:
On further inspection it was found that KLayout was not generating perfectly 45 degree paths.
Therefore, the problem is with KLayout and not open_pdks

closed 2020-12-05 by yrrapt