fossi-foundation/open-pdks #76: sky130_fd_io__top_power_hvc_wpadv2

rsnkhatri3, 2020-12-09:
sky130_fd_io__top_power_hvc_wpadv2 is having the wrong  LEF class(BLOCK instead of PAD). 

So, I  am  unable to  generate padframe,

RTimothyEdwards, 2020-12-09:
@rsnkhatri3 :  Do not use the base cells in a padframe.  This is why they are left as LEF class BLOCK.  What you want to do is to determine which power bus you want the cell to connect to (vddio, vdda, or vccd) and use the appropriate wrapper cell, which is in the sub-library `sky130_ef_io` that is installed with open_pdks.

Although if you want to use a power pad as a straight-through pad with no connections other than ESD diodes, the you might want at least one of the base power pads to be set to a PAD class.  In that case, though, you would want to raise the issue with http://github.com/google/skywater-pdk/.

rsnkhatri3, 2020-12-09:
When using the wrapper cell, sky130_ef_io__vddio_hvc_pad , sky130_fd_io__top_power_hvc_wpad is instantiated in this wrapper cell's verilog module. And while using this sky130_fd_io__top_power_hvc_wpad, it states, "unknown library cell referenced".

RTimothyEdwards, 2020-12-09:
@rsnkhatri3 :  There were references to the deprecated cell `sky130_fd_io__top_power_hvc_wpad` that were apparently missed in the previous purge of deprecated references.  I just corrected them to `sky130_fd_io__top_power_hvc_wpadv2`, which is what they should be.  Thanks for bringing the error to my attention.

RTimothyEdwards, 2020-12-09:
Note that the fix was pushed to the master branch on opencircuitdesign.com and the github mirror will update overnight.

closed 2020-12-15 by rsnkhatri3