fossi-foundation/open-pdks #77: Fake diodes ( sky130_ef_sc_hd__fakediode_2 ) missing SITE in the LEF

smunaut, 2020-12-10:
Looking at the LEF for the fake diode, it's missing the SITE attribute in its LEF.
(I only checked sky130_ef_sc_hd__fakediode_2 but it's possible other ones in other cell libraries have the same issue)

RTimothyEdwards, 2020-12-10:
@smunaut :  That should be the only one.  It's a custom cell with views in open_pdks under `sky130/custom/sky130_fd_sc_hd`,  and it's the only cell there.  Of course, that also highlights the fact that there are no "fake diode" cells for any other standard cell set other than HD, limiting the choices for openlane to resolve antenna violations in any other cell library.

I have updated open_pdks on opencircuitdesign.com with an update to include entries for both SITE and SYMMETRY (which was also missing).

smunaut, 2020-12-10:
I'm actually doing some tests now.

I was always surprised that the diode strategy using fake diodes then replacing them with real ones was always giving worse results. So I now pretty much copied the real diode LEF as the fake diode one, including pad shapes and OBS sections and now I get the exact same results using fake or real diode, which is (1) better in general and (2) what I expect.

So this is what I'm using now :

```
VERSION 5.7 ;
  NOWIREEXTENSIONATPIN ON ;
  DIVIDERCHAR "/" ;
  BUSBITCHARS "[]" ;
MACRO sky130_ef_sc_hd__fakediode_2
  CLASS CORE ;
  FOREIGN sky130_ef_sc_hd__fakediode_2 ;
  ORIGIN 0.000 0.000 ;
  SIZE 0.920 BY 2.720 ;
  SYMMETRY X Y R90 ;
  SITE unithd ;
  PIN DIODE
    DIRECTION INPUT ;
    USE SIGNAL ;
    PORT
      LAYER li1 ;
        RECT 0.085 0.255 0.835 2.465 ;
    END
  END DIODE
  PIN VGND
    DIRECTION INOUT ;
    SHAPE ABUTMENT ;
    USE GROUND ;
    PORT
      LAYER met1 ;
        RECT 0.000 -0.240 0.920 0.240 ;
    END
  END VGND
  PIN VPWR
    DIRECTION INOUT ;
    SHAPE ABUTMENT ;
    USE POWER ;
    PORT
      LAYER met1 ;
        RECT 0.000 2.480 0.920 2.960 ;
    END
  END VPWR
  PIN VPB
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER nwell ;
        RECT -0.190 1.305 1.110 2.910 ;
    END
  END VPB
  PIN VNB
    DIRECTION INOUT ;
    USE GROUND ;
    PORT
      LAYER pwell ;
        RECT 0.145 -0.085 0.315 0.085 ;
    END
  END VNB
  OBS
    LAYER li1 ;
      RECT 0.000000 -0.085000 0.920000 0.085000 ;
      RECT 0.000000  2.635000 0.920000 2.805000 ;
    LAYER mcon ;
      RECT 0.145000 -0.085000 0.315000 0.085000 ;
      RECT 0.145000  2.635000 0.315000 2.805000 ;
      RECT 0.605000 -0.085000 0.775000 0.085000 ;
      RECT 0.605000  2.635000 0.775000 2.805000 ;
  END
END sky130_ef_sc_hd__fakediode_2
END LIBRARY
```

antonblanchard, 2021-01-28:
@smunaut Some fixes just went into `open_pdks` that help with my macros, it might be worth testing again with them.