fossi-foundation/open-pdks #86: Changes to configure script not reflected in documentation

JGrycel, 2020-12-31:
Hi, I'm pretty new to the open EDA community, so let me know if there's a better channel for questions like these. I like OpenLANE's use of open_pdks for the distribution of Sky130, but I've been more successful having my designs build with OpenROAD. So my goal was to use open_pdks standalone to have an up-to-date distribution of Sky130 for use in OpenROAD.

It looks like in commit 8292c90c77b3df28768f182da7dcc7720dd6070c the `--with-sky130-source` option was removed from the config script, but the documentation in both the readme and on [Open PDKs](http://opencircuitdesign.com/open_pdks/index.html) still says to use that option for building. Is the correct new way to specify the location of skywater-pdk to use the `-enable-sky130-pdk` option?

And on a more general level, I'm confused why OpenLANE and OpenROAD are two distinct projects. They seem very similar and differ only in their specific usage of OpenROAD placement, TritonRoute, and PDK distribution. If there's a place where I can learn more about the differences and purposes of each flow that would be helpful.

mithro, 2020-12-31:
> And on a more general level, I'm confused why OpenLANE and OpenROAD are two distinct projects. They seem very similar and differ only in their specific usage of OpenROAD placement, TritonRoute, and [...]

OpenLane and OpenROAD being two distinct projects is more about how history played out rather than an actual goal. As a person involved with both projects, here is my (probably inaccurate) story of how things played out.

At the time OpenLane was started OpenROAD was deep in the middle of trying to show the tooling viable to meet the DARPA IDEA program goals. This required them at the time to focus on the non-open source GlobalFoundries 12nm node (which they succeeded in doing btw!). As you probably understand the type of important problems faced when doing GDS generation from RTL for a 12nm node are a bit different to the problems faced by targeting a 20 year old 130nm process node which leads to some different choices. A concrete example is that OpenROAD only uses KLayout (and Calibre) while SKY130 initially targeting Magic.

To add further complications, open_pdks existed as a way to convert existing proprietary PDKs into a format usable by open tools (targeting multiple proprietary PDKs accessible on the efabless platform). For OpenROAD this conversion was mostly unimportant as they did not need to do DRC checking or spice simulations. Then this was also further confusing for the SkyWater PDK as there is an original proprietary PDK (only available under NDA) that was being initially used with OpenROAD and converted with open_pdks, this is **not** the same as the final PDK that was released (but has some similar characteristics). The current setup for SkyWater PDK also means it is somewhat hard to send pull requests which are mergeable to fix issues, so open_pdks kind of became a place for adding "temporary patches" to work around issues in the upstream SkyWater PDK until it is fixed.

We are still slowly resolving the results, but it is much easier to justify doing that work when there are real users trying to do designs. My current understanding is as follows (but it changes as I get more information);
 * The **aspirational** goal for the SkyWater PDK at https://github.com/google/skywater-pdk is that it can be used directly with tooling without needing open_pdks. 
 * open_pdks will continue to exist and be targeted at allowing people to use proprietary PDKs with open source tools.
 * Now that the OpenROAD team has more time to explore fixing issues with SKY130, OpenLane and OpenROAD teams are trying to figure out the best way forward and meet regularly.

All these things could use help!

JGrycel, 2020-12-31:
Thank you! This makes sense. I was thinking that the open_pdks SKY130 build seems more "standard" in a way, whereas the Sky130 included in OpenROAD seems to just exactly what is needed for building with that tool. But since there (probably?) isn't a standard format for open source libs then maybe neither is "better". The open_pdks version to me seems easier to use and with more integrated configuration files than the OpenROAD distribution.

It seems like the teams on these tools have pretty good collective experience, and I'm not sure if there's any places where I can help out. But if there's any specific areas that need help, especially setup/install and standards for integrating new libs/tools, I'd be interested in contributing. Again, I'm new to this community so I don't have a great feel for how it works.

RTimothyEdwards, 2021-01-01:
@JGrycel :  Posting issues is a great way to start.  Hanging out on the Google/SkyWater slack workspace is also a good way to find out what are the issues that still need solving (if you do not already know where to find it, go to https://join.skywater.tools/).  There are a lot of them.  @mithro :  Thanks for an excellent description of the relationships between openlane and openROAD, and skywater_pdk and open_pdks.

JGrycel, 2021-01-01:
Thanks, sounds good. Is the issue of the `--with-sky130-source` option being removed not a big deal if open_pdks is mainly used to provide Sky130 for OpenLane, and OpenLane rc6 uses an older commit?

RTimothyEdwards, 2021-01-01:
@JGrycel :  The option wasn't removed, it was just renamed to `--enable-sky130-pdk`.

closed 2021-01-02 by JGrycel