fossi-foundation/open-pdks #89: Installation problem with make (and configure script)

britovski, 2021-01-08:
**Get this error as result from 'make' command in installation step 2. Generate files in staging location:**

># Install device subcircuits from vendor files
>set -f ; ../common/foundry_install.py -std_format -source  -target `pwd`/sky130A \
>        -ngspice sky130_fd_pr/latest/models/* \
>        filter=custom/scripts/rename_models.py \
>        |& tee -a sky130A_install.log
>Traceback (most recent call last):
>  File "../common/foundry_install.py", line 428, in <module>
>    sourcedir = option[1]
>IndexError: list index out of range
># Install base device library from vendor files
>set -f ; ../common/foundry_install.py -std_format -source  -target `pwd`/sky130A \
>        -gds %l/latest/cells/*/*.gds compile-only \
>        -cdl %l/latest/cells/*/*.cdl compile-only \
>        -lef %l/latest/cells/*/*.magic.lef compile-only \
>        -spice %l/latest/cells/*/*.spice filter=custom/scripts/rename_cells.py \
>        -library primitive sky130_fd_pr |& tee -a sky130A_install.log
>Traceback (most recent call last):
>  File "../common/foundry_install.py", line 428, in <module>
>    sourcedir = option[1]
>IndexError: list index out of range
># Custom:  Add "short" resistor model and subcircuit to the r+c models file
>cat ./custom/models/short.spice >> \
>        `pwd`/sky130A/libs.tech/ngspice/sky130_fd_pr__model__r+c.model.spice 
>bash: /edatools/DKs/open_pdks/sky130/sky130A/libs.tech/ngspice/sky130_fd_pr__model__r+c.model.spice: No such file or directory
>make[1]: *** [vendor-a] Error 1
>make[1]: Leaving directory `/edatools/DKs/open_pdks/sky130'
>make: *** [all-a] Error 2<

**Previously, the configuration script gave one warning about sources:**

>[root@hera open_pdks]# ./configure --with-sky130-source=/edatools/DKs/skywater-pdk/libraries --with-sky130-local-path=/edatools/DKs
>configure: WARNING: unrecognized options: --with-sky130-source
>configure: Found technology directories: sky130
>configure: Found tools: klayout magic netgen openlane qflow
>checking for magic... /usr/local/bin/magic
>configure: Tools enabled for PDK setup installation: klayout magic netgen openlane qflow
>checking for a Python interpreter with version >= 3.4... python3
>checking for python3... /bin/python3
>checking for python3 version... 3.6
>checking for python3 platform... linux
>checking for python3 script directory... ${prefix}/lib/python3.6/site-packages
>checking for python3 extension module directory... ${exec_prefix}/lib64/python3.6/site-packages
>checking python3 module: distutils... yes
>configure: creating ./config.status
>config.status: creating ../sky130/Makefile
>config.status: creating ../Makefile
>configure: WARNING: unrecognized options: --with-sky130-source
>Build configured successfully

RTimothyEdwards, 2021-01-08:
Note the "WARNING" two lines from the bottom.  The option "--with-sky130-source" was deprecated, and the new replacement option is "--enable-pdk-sky130".  Sorry about the confusion.

britovski, 2021-01-09:
> Note the "WARNING" two lines from the bottom. The option "--with-sky130-source" was deprecated, and the new replacement option is "--enable-pdk-sky130". Sorry about the confusion.

Solved! The correct option is "--enable-sky130-pdk". Thanks.

closed 2021-01-09 by britovski