fossi-foundation/open-pdks #92: ERROR OPTIONS FOR CONFIGURING the SKYWATER-PDK PATH WHEN INSTALLING OPEN-PDK

somefunAgba, 2021-01-15:
```
 sudo ./configure --enable-sky130-pdk=~/projects/foundry/skywater-pdk/libraries --with-sky130-local-path=/cad/pdks
```
_configure: Found technology directories: sky130
configure: Checking specified path for 'sky130' at
checking for ... no
configure: error: Specified path for 'sky130' at  not found_

RTimothyEdwards, 2021-01-15:
The option is "--enable-pdk-sky130", not "--enable-sky130-pdk".

somefunAgba, 2021-01-15:
Okay will try it

somefunAgba, 2021-01-15:
Also,
`sudo ./configure --with-sky130-source=~/projects/skywater-pdk/libraries --with-sky130-local-path=~/cad/pdks`
configure: WARNING: unrecognized options: --with-sky130-source
configure: Found technology directories: sky130
configure: Found tools: klayout magic netgen openlane qflow
checking for magic... /usr/local/bin/magic
configure: Tools enabled for PDK setup installation: klayout magic netgen openlane qflow
checking for a Python interpreter with version >= 3.4... python3
checking for python3... /usr/bin/python3
checking for python3 version... 3.8
checking for python3 platform... linux
checking for python3 script directory... ${prefix}/lib/python3.8/site-packages
checking for python3 extension module directory... ${exec_prefix}/lib/python3.8/site-packages
checking python3 module: distutils... yes
configure: creating ./config.status
config.status: creating ../Makefile
config.status: creating ../sky130/Makefile
**configure: WARNING: unrecognized options: --with-sky130-source**
Build configured successfully

RTimothyEdwards, 2021-01-15:
Yes, the "--with-sky130-source" option was replaced by "--enable-pdk-sky130".

somefunAgba, 2021-01-15:
` sudo ./configure --enable-pdk-sky130=~/projects/foundry/skywater-pdk/libraries --with-sky130-local-path=/cad/pdks
`
**configure: WARNING: unrecognized options: --enable-pdk-sky130**
configure: Found technology directories: sky130
configure: Found tools: klayout magic netgen openlane qflow
checking for magic... /usr/local/bin/magic
configure: Tools enabled for PDK setup installation: klayout magic netgen openlane qflow
checking for a Python interpreter with version >= 3.4... python3
checking for python3... /usr/bin/python3
checking for python3 version... 3.8
checking for python3 platform... linux
checking for python3 script directory... ${prefix}/lib/python3.8/site-packages
checking for python3 extension module directory... ${exec_prefix}/lib/python3.8/site-packages
checking python3 module: distutils... yes
configure: creating ./config.status
config.status: creating ../Makefile
config.status: creating ../sky130/Makefile
**configure: WARNING: unrecognized options: --enable-pdk-sky130**
Build configured successfully

Seems both READMEs have to be updated

somefunAgba, 2021-01-15:
I just cloned repo.

somefunAgba, 2021-01-15:
I just cloned the repo, so I installed the latest version. I am stuck with the unrecognized options error.

RTimothyEdwards, 2021-01-15:
Okay, sorry, this needs debugging.

RTimothyEdwards, 2021-01-15:
All right, I misread the configure script.  Your original configuration line was correct, which I should have known as the correct syntax is given by "configure --help".  Sorry about that confusion.  I looked into this and found that the configuration option does not like "~" in the path name.  I need to add something to the configure script to expand tilde;  but meanwhile, you should be able to get it to work by giving the full absolute path.

somefunAgba, 2021-01-15:
**Thanks. Worked now.**
```
 sudo ./configure --enable-sky130-pdk=/home/somefunagba/projects/foundry/skywater-pdk/libraries --with-sky130-local-path=/cad/pdks
configure: Found technology directories: sky130
configure: Checking specified path for 'sky130' at /home/somefunagba/projects/foundry/skywater-pdk/libraries
checking for /home/somefunagba/projects/foundry/skywater-pdk/libraries... yes
configure: 'sky130' source path found at /home/somefunagba/projects/foundry/skywater-pdk/libraries
configure: Checking whether local path is specified for 'sky130'
configure: Checking whether distribution path is specified for 'sky130'
configure: Option --with-sky130-dist-path=<path> not specified.  Local install only.
configure: Link targets set to none
configure: Found tools: klayout magic netgen openlane qflow
checking for magic... /usr/local/bin/magic
configure: Tools enabled for PDK setup installation: klayout magic netgen openlane qflow
checking for a Python interpreter with version >= 3.4... python3
checking for python3... /usr/bin/python3
checking for python3 version... 3.8
checking for python3 platform... linux
checking for python3 script directory... ${prefix}/lib/python3.8/site-packages
checking for python3 extension module directory... ${exec_prefix}/lib/python3.8/site-packages
checking python3 module: distutils... yes
configure: creating ./config.status
config.status: creating ../Makefile
config.status: creating ../sky130/Makefile
Build configured successfully
```

closed 2021-01-15 by somefunAgba

RTimothyEdwards, 2021-01-25:
FYI, this issue has been completely fixed, and the configure script will now properly expand "~" in a path passed to configure on the command line, in a manner that is cross-platform compatible.

somefunAgba, 2021-01-25:
Thank you. Great work there. Currently using magic and irsim for a course I assist in teaching this semester in Nigeria.
By the way, is there a prm file for processes with lambda = 0.09 and 0.07 micron for magic layouts using Google's skywater-pdk.

RTimothyEdwards, 2021-01-27:
Here is a zip file with IRSIM parameter files for Sky130.  Use the ones with "sky130A" in the name.  The naming convention is sky130A_<voltage>_<temperature>.prm.
 
[sky130_irsim.zip](https://github.com/RTimothyEdwards/open_pdks/files/5881190/sky130_irsim.zip)