fossi-foundation/open-pdks #119: Following instructions fails with magic exit code -9

armleo, 2021-03-28:
Following instructions fails with magic exit code -9
Multiple lines spamming "No label found with index ." in log
It blew my log file to about 50GB so I cant copy and paste it here.

How to fix this?

RTimothyEdwards, 2021-03-29:
I'm looking into it now.  I have not done a full build for a while, and there have been a few changes to the libraries.  I have seen the "No label found" issue before but since it didn't seem to affect the output (and wasn't generating messages to the tune of 50GB) I did not give it high priority.  I can fix that issue but I need to see if there is a deeper underlying problem.

armleo, 2021-03-29:
Hello,

Left it for a night, build both with -j and without (in separate VMs with clean installs of ubuntu with all the packages as specified above) kinda not failed (so it said that it didn't fail, but it might have failed). But I don't know how to verify if everything is compiled correctly.

The log contains magic exit code -9 and "No label found with index ." and it's 10.7 GB in size, no way I can upload that much. I Will try to remove all "No label found with index ." in hopes, it will reduce the size, but GREPing is extremely slow.

Tried opening some xschem projects, assuming that it would install all symbols. Any way to verify this? If I open xschem it fails with scripts/sky130_ something something .tcl not found. I checked the folder it's there but the xschemrc is not the one in git and it does not behave the same way as xschemrc from git. (Specifically, some variables are not set as in xschemrc from git)

I will try to report it on Github of xschem_sky130, later today

Thanks,
Arman

RTimothyEdwards, 2021-03-29:
Don't worry about upload the log.  I can confirm the excessive output from magic;  I am only starting the build but it has run up to 100MB of output, most of which is the "No label found" error.  Any errors you have with xschem startup are most likely to come from open_pdks, not from xschem.

RTimothyEdwards, 2021-03-29:
I have made an update to magic (master branch at opencircuitdesign.com ;  the github repository mirror will be updated overnight tonight) that lets the error messages be suppressed so they won't show up in the output log.  I only got an output log size of about 133MB from a full install; however, the vast majority of that was the error messages from the "readspice" script in magic.  If you can re-run with the updated version of magic and let me know what the output looks like, I'm unsure what else might be wrong, since you were getting a much larger log file output than I got.

armleo, 2021-04-04:
Doing it right now

armleo, 2021-04-04:
Hello,

One more question: Is there any pre-built versions of open-pdks for sky130? Building it takes many hours.

Any recomendations on how to make a pre-built package for ubuntu or arch?

Thanks,
Arman

RTimothyEdwards, 2021-04-04:
@armleo :  All the bits and pieces are still such moving targets that I don't recommend creating a pre-built version, because it will become out of date almost instantly.

The longest part of the build is the "make timing" on the skywater-pdk.  The open_pdks build itself takes maybe half an hour at most, even if all the libraries are installed and enabled.  What I do is to download skywater-pdk separately and do "make timing" and then leave it, and configure open_pdks with "--enable-sky130-pdk=<path>" so that it just picks up those files and does not try to re-do a full build of the PDK.  But the built skywater-pdk files also take up a lot of disk space.  One solution for that is to pre-build the skywater-pdk with only one or two of the standard cell libraries, say HD (area optimized) and HS (speed optimized).  That considerably cuts down on the build time.  You can also disable the build of the SRAM macro in open_pdks, depending on whether or not you want the SRAM macro (eventually, the full public release of OpenRAM with sky130 gets published and I'll remove the macro build from open_pdks).

armleo, 2021-04-04:
timing takes like on hour, it's open_pdks are taking like many many hours

armleo, 2021-04-17:
Looks like if i disable sram macro cell it will build in couple of minuts, but with it, it takes really long time and spams the messages.
Currently openlane contains different version for magic, so I am unable to test it with latest magic.

RTimothyEdwards, 2021-04-18:
Maybe I should disable the SRAM by default, at least for now.

closed 2021-07-24 by armleo