fossi-foundation/open-pdks #121: configure: error: You need 'magic' to generate the needed various cell views.

Kreijstal, 2021-05-16:
just git cloned and did `./configure --enable-sky130-pdkc` 
just as the [manual](http://opencircuitdesign.com/open_pdks/) says, maybe there is something else that should be added to the documentation? Or is this a bug?

Kreijstal, 2021-05-16:
Ah, vieweing [this](https://github.com/0x01be/openpdks-docker/blob/timedwards/Dockerfile) docker file put some insight in what I was doing wrong, apparently you have a dependency on magic. I see that it's mentioned that some packages might have a dependency on magic, it wasn't as clear though, thank you anyway.

closed 2021-05-16 by Kreijstal

ximinsiwuya, 2024-02-27:
May I ask how to solve the problem about "magic"? Thank you very much for your suggestion

Kreijstal, 2024-02-27:
https://github.com/silicon-vlsi-org/eda-magic/blob/main/ADMIN.md

RTimothyEdwards, 2024-02-27:
@ximinsiwuya :  Or clone it from https://github.com/RTimothyEdwards/magic and compile and install according to the instructions at http://opencircuitdesign.com/magic/.