fossi-foundation/open-pdks #126: Trying to create `/usr/share/pdk` when `--with-sky130-local-path` set to different location.

mithro, 2021-06-13:
```
Done with PDK migration.
echo "Ended SKY130 PDK migration on "`date` >> sky130A_install.log
make[3]: Leaving directory '/ssd/gob/foss-eda-tools/skywater-pdk-libs-sky130_fd_bd_sram/third_party/open_pdks/sky130'
make[2]: Leaving directory '/ssd/gob/foss-eda-tools/skywater-pdk-libs-sky130_fd_bd_sram/third_party/open_pdks/sky130'
mkdir -p /usr/share/pdk/bin/
mkdir: cannot create directory ‘/usr/share/pdk’: Permission denied
make[1]: *** [Makefile:86: common_install] Error 1
make[1]: Leaving directory '/ssd/gob/foss-eda-tools/skywater-pdk-libs-sky130_fd_bd_sram/third_party/open_pdks'
make: *** [/ssd/gob/foss-eda-tools/skywater-pdk-libs-sky130_fd_bd_sram/scripts/make/pdk.mk:56: /ssd/gob/foss-eda-tools/skywater-pdk-libs-sky130_fd_bd_sram/env/conda/envs/sky130_fd_bd_sram/share/pdks/sky130A/.stamp] Error 2
```

mithro, 2021-06-13:
It feels like you should just be able to do something like;
```diff
diff --git a/scripts/configure b/scripts/configure
index 80d66e5..659b571 100755
--- a/scripts/configure
+++ b/scripts/configure
@@ -585,7 +585,6 @@ PACKAGE_STRING='open_pdks 1.0'
 PACKAGE_BUGREPORT='github.com/RTimothyEdwards/open_pdks'
 PACKAGE_URL=''
 
-ac_default_prefix=/usr/share
 ac_subst_vars='LTLIBOBJS
 LIBOBJS
 EF_STYLE
@@ -2205,7 +2204,7 @@ $as_echo "$as_me: Found technology directories: sky130" >&6;}
 
 
         SKY130_SOURCE_PATH=""
-        SKY130_LOCAL_PATH="/usr/share/pdk"
+        SKY130_LOCAL_PATH="${datarootdir}/pdk"
         SKY130_DIST_PATH=""
         SKY130_LINK_TARGETS="none"
         SKY130_AUTO="0"
diff --git a/scripts/configure.ac b/scripts/configure.ac
index a4f07d2..36c4dcf 100755
--- a/scripts/configure.ac
+++ b/scripts/configure.ac
@@ -1,6 +1,6 @@
 AC_INIT([open_pdks], [1.0], [github.com/RTimothyEdwards/open_pdks])
 
-AC_PREFIX_DEFAULT(/usr/share)
+AC_PREFIX_DEFAULT(/usr/local)
 
 ## REQUIRES: https://www.gnu.org/software/autoconf-archive/ax_python_module.html#ax_python_module
 AM_PATH_PYTHON([3.4])
@@ -36,7 +36,7 @@ AC_DEFUN([M4_GEN_WITH_PDK_ARGS], [
         m4_define([pdkvar], [m4_normalize(m4_esyscmd(echo pdk | tr "a-z-" "A-Z_"))])
 
         pdkvar[]_SOURCE_PATH=""
-        pdkvar[]_LOCAL_PATH="/usr/share/[pdk]"
+        pdkvar[]_LOCAL_PATH="${datarootdir}/[pdk]"
         pdkvar[]_DIST_PATH=""
         pdkvar[]_LINK_TARGETS="none"
         pdkvar[]_AUTO="0" 
```

mithro, 2021-06-13:
Fixed by #127

RTimothyEdwards, 2021-08-22:
#127 has been pulled and merged, so closing this issue.

closed 2021-08-22 by RTimothyEdwards