fossi-foundation/open-pdks #128: "IndexError: list index out of range" not causing install to fail

mithro, 2021-06-14:
I'm seeing errors like the following but the install doesn't seem to be failing;
```
# Install OSU T18 ls digital standard cells.
set -f ; ../common/foundry_install.py -std_format -source  -target /ssd/gob/foss-eda-tools/skywater-pdk-libs-sky130_fd_bd_sram/third_party/open_pdks/sky130/sky130A \
        -techlef sky130_osu_sc_18T.tlef rename=sky130_osc_sc_18t_ls.tlef \
        -spice 18T_ls/cdl/*.spice compile-only exclude=*.*.spice \
        -lef 18T_ls/lef/*.lef \
        -lib 18T_ls/lib/*.lib \
        -gds 18T_ls/gds/*.gds compile-only \
                options=custom/scripts/gds_import_setup.tcl \
        -library digital sky130_osu_sc_18t_ls 2>&1 | tee -a sky130A_make.log
Traceback (most recent call last):
  File "../common/foundry_install.py", line 430, in <module>
    sourcedir = option[1]
IndexError: list index out of range
```

apond308, 2021-07-07:
Also having this issue. Any tips would be great!

RTimothyEdwards, 2021-07-08:
It's not a fatal issue in any way, but there are parts of the Makefile that should be completely ignoring libraries that have not been enabled during configure, and instead it's attempting to install them.  Since it's not affecting the rest of the install, I've put the issue on the back burner until I have some time to look into it (along with a lot of other unnecessarily alarming error and warning messages).

mithro, 2021-11-25:
I believe this was fixed by https://github.com/RTimothyEdwards/open_pdks/pull/186

There is also #189 which takes it the next step further.

closed 2021-11-25 by mithro