fossi-foundation/open-pdks #130: `wire 1;`

dan-fritchman, 2021-06-17:
A few gate-level models have Verilog-illegal wire-name-declarations, generally named `wire 1;`. 

(Apologies if these are in the skywater-pdk source material, but I couldn't find them in there.) 

From `sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v` (notable line commented): 

```Verilog 
// Import user defined primitives.
`celldefine
module sky130_fd_sc_hd__dlxbn (
    Q     ,
    Q_N   ,
    D     ,
    GATE_N,
    VPWR  ,
    VGND  ,
    VPB   ,
    VNB
);

    // Module ports
    output Q     ;
    output Q_N   ;
    input  D     ;
    input  GATE_N;
    input  VPWR  ;
    input  VGND  ;
    input  VPB   ;
    input  VNB   ;

    // Local signals
    wire GATE          ;
    wire buf_Q         ;
    wire GATE_N_delayed;
    wire D_delayed     ;
    reg  notifier      ;
    wire awake         ;
    // wire 1             ; FIXME! This is NOT a comment in the real version 

    //                                    Name     Output  Other arguments
    not                                   not0    (GATE  , GATE_N_delayed                       );
    sky130_fd_sc_hd__udp_dlatch$P_pp$PG$N dlatch0 (buf_Q , D_delayed, GATE, notifier, VPWR, VGND);
    assign awake = ( VPWR === 1 );
    buf                                   buf0    (Q     , buf_Q                                );
    not                                   not1    (Q_N   , buf_Q                                );

endmodule
`endcelldefine
```

RTimothyEdwards, 2021-06-17:
Yes, they are in the skywater-pdk source material, here:  `skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/dlxbn/sky130_fd_sc_hd__dlxbn.behavioral.v`.  There is an issue for it in the skywater-pdk repository.  I can add a patch for this one in open_pdks, though.  But there are some other ones that are more complicated and best handled in the skywater-pdk repository itself.

dan-fritchman, 2021-06-17:
OK, thanks for pointing that out, and agree it belongs over there.