fossi-foundation/open-pdks #143: Qflow GUI

buraktoker, 2021-08-08:
Hello everyone,

I am newbie to Qflow and open source VLSI flow. I have a question about Qflow and this pdk. 

Is there any way to see and use these libraries in qflow_gui "Preparation" part like OSU_035 ?

RTimothyEdwards, 2021-08-09:
The best way is to have a symbolic link called "tech" in the same directory as you create a project that is a link to the qflow setup in sky130A;  e.g., for the default install, `ln -s /usr/share/pdk/sky130A/libs.tech/qflow tech`.  Make this symbolic link before you run `qflow gui`, and you should be able to see all of the digital standard cell libraries in the qflow GUI.

A word of warning:  The high-density standard cells do not work very well with qflow;  it will generate a lot of DRC errors.  Use the "HS", "MS", and "LS" libraries to minimize the number of DRC issues (I have not had the time to debug some of the more common problems, but the number of DRC errors is manageable for small-ish designs).

buraktoker, 2021-08-10:
Hello, 
Thanks a lot for your reply.
However, when I used openpdk libs synthesis gives me an error. Synth.log is shown below.
If I used OSU018,this error didn't happen.

Qflow synthesis logfile created on Sal Ağu 10 19:49:45 +03 2021
Running yosys for verilog parsing and synthesis
yosys  -s fulladder.ys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7 (git sha1 61f6811, gcc 6.2.0-11ubuntu1 -O2 -fdebug-prefix-map=/build/yosys-OIL3SR/yosys-0.7=. -fstack-protector-strong -fPIC -Os)

-- Executing script file `fulladder.ys' --

1. Executing Liberty frontend.
Imported 376 cell types from liberty file.

2. Executing Verilog-2005 frontend.
Parsing Verilog input from `/home/burakt/Desktop/ASIC/adder2/source/adder.v' to AST representation.
Generating RTLIL representation for module `\fulladder'.
Generating RTLIL representation for module `\rippe_adder'.
Successfully finished Verilog frontend.

3. Executing SYNTH pass.

3.1. Executing HIERARCHY pass (managing design hierarchy).

3.1.1. Analyzing design hierarchy..
Top module:  \fulladder

3.1.2. Analyzing design hierarchy..
Top module:  \fulladder
Removing unused module `\rippe_adder'.
Removed 1 unused modules.

3.2. Executing PROC pass (convert processes to netlists).

3.2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

3.2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

3.2.3. Executing PROC_INIT pass (extract init attributes).

3.2.4. Executing PROC_ARST pass (detect async resets in processes).

3.2.5. Executing PROC_MUX pass (convert decision trees to multiplexers).

3.2.6. Executing PROC_DLATCH pass (convert process syncs to latches).

[... 415 lines cut ...]
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrbn_1".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrbn_2".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrbp_1".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrbp_2".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrtn_1".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrtn_2".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrtn_4".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrtp_1".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrtp_2".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlrtp_4".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlxbn_1".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlxbn_2".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlxbp_1".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlxtn_1".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlxtn_2".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlxtn_4".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "sky130_fd_sc_hs__dlxtp_1".
ABC: Scl_LibertyReadGenlib() skipped three-state cell "sky130_fd_sc_hs__ebufn_1".
ABC: Scl_LibertyReadGenlib() skipped three-state cell "sky130_fd_sc_hs__ebufn_2".
ABC: Scl_LibertyReadGenlib() skipped three-state cell "sky130_fd_sc_hs__ebufn_4".
ABC: Scl_LibertyReadGenlib() skipped three-state cell "sky130_fd_sc_hs__ebufn_8".
ABC: Scl_ERROR: ABC: execution of command "/usr/local/share/qflow/bin/yosys-abc -s -f /tmp/yosys-abc-jb9MQs/abc.script 2>&1" failed: return code 134.
outputprep failure:  No file fulladder_mapped.blif.
Premature exit.
Synthesis flow stopped due to error condition.

RTimothyEdwards, 2021-08-25:
@buraktoker :  I think your main problem is that you're using qflow version 1.3 which is completely deprecated and unsupported.