fossi-foundation/open-pdks #151: I did not ask for osu library only sky but there is a file causing havoc

svenn71, 2021-08-24:
Somehow the osu lib seems to be injected.
I only did ./configure --enable-sky130-pdk

sky130/sky130A/libs.tech/qflow/sky130Aosu.sh is empty and cause problem

```
FileNotFoundError: [Errno 2] No such file or directory: 'qflow/sky130_osu_sc_t18.sh'
make[2]: *** [Makefile:625: qflow-a] Error 1
make[2]: Leaving directory '/home/svennb/work/prg/git/open_pdks/sky130'
make[1]: *** [Makefile:562: all-a] Error 2
make[1]: Leaving directory '/home/svennb/work/prg/git/open_pdks/sky130'
make: *** [Makefile:103: tech_sky130] Error 2
```

RTimothyEdwards, 2021-08-24:
Oops, I never pushed that change.  I just pushed it a few minutes ago, so please do a git pull and do the "configure" again.  Thanks for bringing it to my attention.

svenn71, 2021-08-24:
the issue now fixed

closed 2021-08-24 by svenn71