fossi-foundation/open-pdks #159: Resistor models don't work.

armleo, 2021-09-10:
When creating instance of resistor, NGSPICE complains that resistor model has not been found

RTimothyEdwards, 2021-09-11:
Which resistor model, and how are you including the SPICE model libraries?  Please include a repeatable test case.

RTimothyEdwards, 2021-09-11:
Hopefully my response on the #analog-design channel in the skywater-pdk slack workspace is what you're looking for?

armleo, 2021-09-11:
NGSPICE Log:
```
ngspice -r "serdes/resisor_issue.raw" serdes/resisor_issue.spice
******
** ngspice-34 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2020, The ngspice team.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Sat Mar  6 03:53:35 UTC 2021
******

Compatibility modes selected: hs a

Circuit: **.subckt resisor_issue

Error: unknown subckt: xr1 0 net1 sky130_fd_pr__res_generic_m1 w=1 l=1 mult=1 m=1
    Simulation interrupted due to error!

ngspice 1 -> 
```

Netlist (my install path is /opt/pdk_root, other models like transistors work):
```
**.subckt resisor_issue
V2 net1 GND vdd_volts
XR1 GND net1 sky130_fd_pr__res_generic_m1 W=1 L=1 mult=1 m=1
**** begin user architecture code

.temp 125
.param vdd_volts=1.65
.param vddio_volts=3.0

.control
tran 0.01ns 20ns
write
.endc

.param mc_mm_switch=0

.lib "/opt/pdk_root/sky130A/libs.tech/ngspice/sky130.lib.spice" ss

**** end user architecture code
**.ends
.GLOBAL GND
** flattened .save nodes
.end
```

armleo, 2021-09-11:
Hmm... Maybe it's not a .subckt but resistor model and I need to use RR1 instead of XR1?

armleo, 2021-09-11:
Update: The X-> R replacement in SPICE helped, but now it gives `mult` parameter not found, the netlist is generated from xschem using the XSCHEM symbols from Open_PDKs, they may need to be updated to remove that parameter.

armleo, 2021-09-11:
Could you please apply the changes or do I need to report it somewhere else?

armleo, 2021-09-11:
Issue has been solved in xschem_sky130 and pushed to upstream. Closing this now. (I am assuming the open_pdks will just install latest xschem_sky130)

closed 2021-09-11 by armleo