fossi-foundation/open-pdks #162: LU factorization of the G Matrix failed

peijunh, 2021-09-18:
Hello. I want to thank you for the update on common_pdn.tcl. After the update of @Manarabdelaty , I run my design with macro and get the error in the following figure. 

[image]

My macro includes a sensor and sky130_sram_4kbyte_1rw1r_32x1024_8. sensor use VDD and VSS as power pin and sram use vccd1 and vssd1 as its power pin. So I change the "set macro" in common.tcl as follows. 
>set macro {
>    orient {R0 R180 MX MY R90 R270 MXR90 MYR90}
>    power_pins "vccd1 VPWR VDD"
>    ground_pins "vssd1 VGND VSS"
>    blockages "li1 met1 met2 met3 met4"
>    straps {
>    }

Could you share suggestion and advice on this error? Thanks!

getziadz, 2021-09-18:
I have encountered the same issue after the latest update.

RTimothyEdwards, 2021-09-18:
Please contact @Manarabdelaty , who was last working on the PDN code.  I don't really have much to do with the openlane development;  I just get pull requests for updates and merge them.  The best place to get a fast response is the skywater-pdk slack workspace, #openlane channel.

peijunh, 2021-09-20:
@getziadz I think we can try to set FP_PDN_IRDROP to 0 in config.tcl to avoid this.

closed 2021-09-20 by peijunh

getziadz, 2021-09-20:
Yes. That worked for me too.