fossi-foundation/open-pdks #163: ngspice simulation start very slow

hpretl, 2021-09-24:
The way the SPICE model files are organized in sky130A seem to cause a long startup time of ngspice due to the read of the model files. Is there a way to speed this up? Maybe a single, flat model file w/o the many includes would be faster?

RTimothyEdwards, 2021-09-24:
Are you copying the file `/usr/share/pdk/sky130A/libs.tech/ngspice/spinit` to your local simulation directory and renaming it to `.spiceinit`?

If that is still not fast enough for you, please complain to the ngspice developers (namely Holger Vogt) that the time to parse the file passed to .lib should be proportional to the number of entries in one corner section, not proportional to the number of corner sections in the file, and that this should be considered a parsing error that needs to be fixed.

However, in general, once I use the .spiceinit, the startup time, while long, is tolerable.  It depends on whether you are talking about 30 seconds startup time, or 30 minutes.

You might also want to try to convince Holger that the settings in the `spinit` file are so obviously normal that they should be defaults.  But that might be a harder sell.

hpretl, 2021-09-24:
Thanks for the hint with `spinit`, I think it helped a bit but not much.

I ran a little speed test on my Ubuntu machine (2 trials each to exclude first-run effects, time from simulation start in `xschem` to actual simulation run in `ngspice`):

Using `sky130.lib.spice` as is: **85s**
Commenting everything but the `tt` section: **30s**

So there seems to be a factor 3x potential by skipping the unused model sections in `ngspice` (I will try to contact them, maybe they can improve this), but even the 30s seems long. If I find some time I will try to flatten the model file, and see how much that brings. With another factor 3x I am down to 10s, this would be ideal.

hpretl, 2021-10-03:
I have written a small Python script that (1) extracts only the given corner, (2) removes comments and empty lines, and (3) produces a single flat model file.

On my machine, this improves `ngspice` time from start of simulator to actual simulation start from **80sec to 5sec**!

It can be found here: [https://github.com/hpretl/spice_model_red](https://github.com/hpretl/spice_model_red)

If you want to include this into `open_pdks`, please feel free, I think it would be quite handy for most of the people.

RTimothyEdwards, 2021-10-04:
Thank you!  I have added the script to the runtime/ directory, which will install it with the other run-time scripts.

hpretl, 2021-10-04:
Cool, thanks! I am happy to being able to contribute a tiny bit to this great initiative.

RTimothyEdwards, 2021-10-04:
Thanks for the heads-up.  I have had issues with the PNP models which were supposed to be corrected by patches when installing open_pdks, but they have been known to break every now and then.  I'll look into it.

RTimothyEdwards, 2021-10-10:
I wouldn't necessarily recommend closing this ticket as you have mentioned the PNP model issue.  Maybe that should be opened as a separate ticket, since it's a separate topic?  I haven't had the chance to look into it yet.

hpretl, 2021-10-10:
When `spice_model_red` is run on the current OpenLane version (2021.09.30_02.12.16), after the `make pdk`, it flags the following include files as missing:

Warning! File sky130_fd_pr__pnp_05v5_w3p40l3p40.model.spice not found.
Warning! File sky130_fd_pr__pnp_05v5_w0p68l0p68.model.spice not found.

armleo, 2021-12-11:
I have this issue.
I have been able to optimize the .lib.spice to only contain the .lib-s of the specific corner by separating it to multiple pieces.
Results: NGSPICE runtime reduced from 500 seconds to a couple of seconds.

RTimothyEdwards, 2021-12-11:
@armleo :  This is specfically an ngspice problem and should be raised as an issue with the ngspice developers.  There is no reason whatsoever that ngspice should process anything outside of the `.lib`..`.endl` block that corresponds to the corner name passed to ".lib".  That additional unrelated blocks increase processing time indicates that the ngspice parser is not simply treating the `.lib` statements like an "ifdef" block for a pre-processor but is apparently reading and processing every block in the file before applying the one that is selected.  I see nothing in the description of the `.lib` statement in the ngspice documentation that would suggest that there is any utility in processing statements in the `.lib` file that are unrelated to the selected block.

armleo, 2021-12-11:
As far as I am aware spice parsing is heavily context based, so just ignoring everything in .lib/.endl should not be possible (at least is not easy).

I think we can mitigate this by creating multiple .lib.spices corresponding to each lib, right?

hpretl, 2021-12-11:
> I have this issue. I have been able to optimize the .lib.spice to only contain the .lib-s of the specific corner by separating it to multiple pieces. Results: NGSPICE runtime reduced from 500 seconds to 5 seconds.
> 
> Any suggestions on how to upstream this?

You can use my script in https://github.com/hpretl/spice_model_red to prepare flat SPICE model files per corner, which read in much faster in `ngspice`.

hpretl, 2021-12-28:
Update: I have moved and renamed the `spice_model_red`, it is now part of [https://github.com/hpretl/iic-osic](https://github.com/hpretl/iic-osic), and can be found here: [iic-spice-model-red](https://github.com/hpretl/iic-osic/blob/main/iic-spice-model-red.py).

closed 2023-02-10 by hpretl