fossi-foundation/open-pdks #173: make install with DESTDIR doesn't work

mithro, 2021-11-13:
Seems like `-local` argument to `../common/staging_install.py` doesn't work.
```
+ cd open_pdks
+ ./configure --enable-sky130-pdk=/host/skywater-pdk --enable-alpha-sky130 --enable-xschem-sky130 --prefix=/usr/local
checking for a Python interpreter with version >= 3.4... python3
checking for python3... /usr/bin/python3
checking for python3 version... 3.9
checking for python3 platform... linux
checking for python3 script directory... ${prefix}/lib/python3.9/site-packages
checking for python3 extension module directory... ${exec_prefix}/lib/python3.9/site-packages
checking python3 module: distutils... yes
checking for a sed that does not truncate output... /bin/sed
configure: Found technology directories: sky130
Checking technology sky130...
configure: Checking specified path for 'sky130' at /host/skywater-pdk
checking for /host/skywater-pdk... yes
configure: 'sky130' source path found at /host/skywater-pdk
configure: Link targets set to none
configure: Found tools: klayout magic netgen irsim openlane qflow xschem
checking for magic... /host/out/magic/bin/magic
checking for patch... /usr/bin/patch
configure: Package 'sky130_ml_xx_hd' will be installed automatically during make.
configure: Package 'xschem_sky130' will be installed automatically during make.
configure: Package 'sky130_sram_macros' will not be installed.
configure: Package 'sky130_osu_t12' will not be installed.
configure: Package 'sky130_osu_t15' will not be installed.
configure: Package 'sky130_osu_t18' will not be installed.
configure: Tools enabled for PDK setup installation: klayout magic netgen irsim openlane qflow xschem
configure: creating ./config.status
config.status: creating ../sky130/Makefile
config.status: creating ../Makefile
Build configured successfully
+ make DESTDIR=/host/out/pdk install
(cd sky130 && make install)
make[1]: Entering directory '/host/open_pdks/sky130'
echo "Starting SKY130 PDK migration on "`date` > sky130A_install.log
../common/staging_install.py -local /host/out/pdk -std_format \
        -source /host/open_pdks/sky130/sky130A \
        -target /usr/local/share/pdk/sky130A \
        -variable PDKPATH \
        -link_from none 2>&1 | tee -a sky130A_install.log
Installing in target directory /usr/local/share/pdk/sky130A
Removing files from target
Copying staging files to target
Done.
Changing local path references from /host/open_pdks/sky130/sky130A to /host/out/pdk
Part 1:  Tools
      xcircuit (1 substitution)
      qflow (16 substitutions)
      xschem (1 substitution)
      magic (2 substitutions)
Part 2:  Libraries
   sky130_fd_sc_hvl
      mag (1 substitution)
      maglef (1 substitution)
   sky130_fd_pr
      mag (1 substitution)
      maglef (1 substitution)
   sky130_fd_io
      mag (1 substitution)
      maglef (1 substitution)
   sky130_ml_xx_hd
   sky130_fd_sc_hd
      mag (1 substitution)
      maglef (1 substitution)
Removing temporary files from destination.
Done with PDK migration.
echo "Ended SKY130 PDK migration on "`date` >> sky130A_install.log
make[1]: Leaving directory '/host/open_pdks/sky130'
Common install:  Done.
Done.
+ ls -l /host/out
total 4
drwxr-xr-x 5 root root 4096 Nov 13 02:57 magic
```

RTimothyEdwards, 2021-11-13:
@mithro :  Driver error?  "-local" simply changes all of the references in the files to point to the indicated path.  The files are still installed in "target".  The "local" option is meant to handle what I do on the efabless platform, which is to set "target" to the git repo that holds the PDK, which is something like ~/gits/skywater-pdk, and set "local" to the /ef/ tree location /ef/tech/SW.2/.  Then when I push the git repo, there's a server that distributes it to all the hosts, where it gets installed in /ef/tech/SW.2/.  So all the path references need to be set to the final install location, not the target.  Possibly you have "local" and "target" reversed?

mithro, 2021-11-13:
I'm not calling `../common/staging_install.py` directly, so it seems like an issue in your `./configure` and `Makefile`?

The line `./configure --enable-sky130-pdk=/host/skywater-pdk --enable-alpha-sky130 --enable-xschem-sky130 --prefix=/usr/local` should mean I'm generating output which will eventually be installed under `/usr/local`

The line `make DESTDIR=/host/out/pdk install` is suppose to put the files under the directory `/host/output/pdk` in a form they can eventually be installed under `/usr/local` through a packaging system or similar.

mithro, 2021-11-13:
This is described in https://github.com/RTimothyEdwards/open_pdks/pull/127

mithro, 2021-11-13:
Also, according to the documentation at the top of `common_install.py`,

 *  ​`-target` ​should be final resting place of the PDK
 * `-local` should be the actual file system location where the files get written too.

```
 ​-target <path>     Final install path in system file system.

                    ​Normally, '$(prefix)/pdks/<unique pdk name>'.

                    ​If -local is not given, this will be the top level
                    ​directory location the files are installed too.

 ​-local <path>      Actual file system location to write the files too.
                    ​The result can then be packaged and distributed.

                    ​For usage with things like package managers and other
                    ​administrator installation tooling.  The resulting
                    ​files still need to be installed at '-target' on the
                    ​final system.

                    ​Think 'DESTDIR', see
                    ​https://www.gnu.org/prep/standards/html_node/DESTDIR.html
```

mithro, 2021-11-13:
> @mithro : Driver error? "-local" simply changes all of the references in the files to point to the indicated path. The files are still installed in "target". The "local" option is meant to handle [...]

What you are doing on the described above is pretty similar to what I'm trying to do. 

What you describe would be done with `./configure --prefix=/ef/tech/SW.2/` (the final "on system" destination is `/ef/tech/SW.2`) and `make DESTDIR=~/gits/skywater-pdk install` (the location to write the files for distribution to systems is `~/gits/skywater-pdk`).

RTimothyEdwards, 2021-11-13:
@mithro:  All the stuff in the Makefile related to DESTDIR is the stuff that you put there.

mithro, 2021-11-13:
How are you currently getting your "efabless behaviour"?

RTimothyEdwards, 2021-11-13:
Mostly "efabless behavior" just swaps the library and file-type directories (e.g., "sky130_fd_sc_hd/lef" vs. "lef/sky130_fd_sc_hd"), adds an extra directory "current" after "libs.tech/magic", and a few other idiosyncrasies that keep backwards compatibility with the scripts and file system on the efabless platform.

mithro, 2021-11-13:
I meant how are you doing the efabless install procedure?

RTimothyEdwards, 2021-11-13:
I was able to get it to work by doing `make DESTDIR=/ef/tech/SW.2 SHARED_PDKS_PATH=/home/tim/gits/ef-skywater-sky130 install`

That installs the PDK into my own local git repo where I can then push it to trigger the system update, but all the references in the files point to the final destination in the /ef/tech/ path.

mithro, 2021-11-14:
To fix this, do you want me to modify the python script of the Makefiles?

RTimothyEdwards, 2021-11-14:
Whichever one makes the most sense.  My impression was that DESTDIR and SHARED_PDKS_PATH should be swapped, which would indicate that the solution should be in the Makefile or configure script.  As long as the default behavior is to install in /usr/local/share/pdk and there is some way to specify the install so that it works the way I need it to on the efabless platform, I'm okay with it.

mithro, 2021-11-15:
I believe I have a fix in #175.

olofk, 2022-01-25:
I see #175 has been merged. Can we close this?