fossi-foundation/open-pdks #177: ./configure --disable-XXX doesn't seem to work

mithro, 2021-11-15:
```
./configure --enable-sky130-pdk=/host/skywater-pdk --enable-alpha-sky130 --prefix=/usr/local --enable-klayout --disable-magic --disable-netgen --disable-irsim --disa
ble-openlane --disable-qflow --disable-xschem                                                                                                                          
checking for a Python interpreter with version >= 3.4... python3                                                                                                       
checking for python3... /usr/bin/python3                                                                                                                               
checking for python3 version... 3.9                                                                                                                                    
checking for python3 platform... linux                                                                                                                                 
checking for python3 script directory... ${prefix}/lib/python3.9/site-packages                                                                                         
checking for python3 extension module directory... ${exec_prefix}/lib/python3.9/site-packages                                                                          
checking python3 module: distutils... yes                                                                                                                              
checking for a sed that does not truncate output... /bin/sed                                                                                                           
configure: Found technology directories: sky130                                                                                                                        
Checking technology sky130...                                                                                                                                          
configure: Checking specified path for 'sky130' at /host/skywater-pdk                                                                                                  
checking for /host/skywater-pdk... yes                                                                                                                                 
configure: 'sky130' source path found at /host/skywater-pdk                                                                                                            
configure: Link targets set to none                                                                                                                                    
configure: Found tools: klayout magic netgen irsim openlane qflow xschem
checking for magic... /host/out/magic/bin/magic
checking for patch... /usr/bin/patch
configure: Package 'sky130_ml_xx_hd' will be installed automatically during make.
configure: Package 'xschem_sky130' will be installed automatically during make.
configure: Package 'sky130_sram_macros' will not be installed.
configure: Package 'sky130_osu_t12' will not be installed.
configure: Package 'sky130_osu_t15' will not be installed.
configure: Package 'sky130_osu_t18' will not be installed.
configure: Tools enabled for PDK setup installation: klayout magic netgen irsim openlane qflow xschem
configure: creating ./config.status
config.status: creating ../sky130/Makefile
config.status: creating ../Makefile
Build configured successfully
```

mithro, 2021-11-15:
Looking at the following code in the `configure` script;
```
        KLAYOUT_DISABLED=0
        # Check whether --enable-klayout was given.
if test "${enable_klayout+set}" = set; then :
  enableval=$enable_klayout;
                if test "$enableval" == "no" -o "$enableval" == "NO"; then
                    KLAYOUT_DISABLED=1
                fi

fi
```
I also tried `--enable-XXXX=no` but it also had no effect.

mithro, 2021-11-15:
Actually it looks like the `Tools enabled for PDK setup installation:` line is hard coded?
https://github.com/RTimothyEdwards/open_pdks/blob/4768d03a9fb22a8e453ccf40aa3f3216f08973df/scripts/configure#L2735-L2736

mithro, 2021-11-15:
Looks like it comes from here;
https://github.com/RTimothyEdwards/open_pdks/blob/4768d03a9fb22a8e453ccf40aa3f3216f08973df/scripts/configure.ac#L214-L234

mithro, 2021-11-15:
As far as I can see, the `--enable-tool` / `--disable-tool` has no effect on the files produced.

Park-Hyung-Joo, 2022-03-11:
[image]
I have another kind of issue as you can see on this pic, --disable-XXX-sky130 occurs index error and xschem source is made

RTimothyEdwards, 2022-03-11:
@Park-Hyung-Joo :  I can confirm the "Traceback" message from the Configure script.  However, when I ran the build after configuring with `--disable-xschem-sky130` and `--disable-alpha-sky130`, those packages were not cloned, and the corresponding PDK directories were not populated and installed, as expected.

itsayellow, 2025-01-24:
Yes I'm just running into this.  All the `--disable-<tool>` or `--enable-<tool>=no` options seem to have no effect.

I can accept this reality, however it would at least be nice not to list them with `configure --help` if they do nothing.

RTimothyEdwards, 2025-01-25:
@itsayellow :  In the original issue, I determined that the options _did_ have the expected effect, even though a python error was raised.  But you are reporting that the options have _no_ efffect, which is new.  Are you seeing this for all tools, or just hit one in particular that failed?

itsayellow, 2025-01-27:
Thanks for the tip @RTimothyEdwards.

I was having full disk troubles, so I can try it again and verify.  But it appeared that the switches for tools weren't working or doing anything:

```shell
  --enable-klayout=no
  --enable-magic=no
  --enable-netgen=no
  --enable-irsim=no
  --enable-openlane=no
  --enable-qflow=no
  --enable-xschem=no
  --enable-xcircuit=no
```
Do these only have an effect during `make install`?  That could explain why everything appeared the same during `make`.

Because my disk was filling up, I had to get more disk space.  When I finally did get `./configure && make && make install` to work, I omitted all of those switches.  So I will try again and see if I get a different result with some of those switches set to `--enable-<tool>=no`

RTimothyEdwards, 2025-01-27:
@itsayellow :  I will certainly check on why the configuration options don't work.  However, as a solution to your problem, you want to take a different tack instead.  The tools install files that take up only a tiny amount of disk space compared to the PDK libraries.  The tool files are mostly just scripts and setup files.  Your better bet is to disable unused libraries.  The standard cell libraries can be huge, and many of the standard cell sets have not even been properly integrated into the place and route tools.  I suggest retaining `sky130_fd_sc_hd` and `sky130_fd_sc_hvl` and disbling all of the other `sky130_fd_sc_*` libraries, which will save you a lot of disk space.

The other option is to use https://github.com/efabless/volare, which is basically a package manager for the open PDKs.  That avoids all the trouble and overhead of building the PDK yourself.

itsayellow, 2025-01-27:
Thanks for all the tips!  They are really useful.

I remember now why it seemed like the `--enable-<tool>` didn't work.  With the following command:

```shell
./configure --enable-klayout=no --enable-xcircuit=no --enable-netgen=no --enable-irsim=no --enable-openlane=no --enable-qflow=no
```

I get the following output
```
[...]
configure: Tools enabled for PDK setup installation: klayout magic netgen irsim openlane qflow xschem xcircuit
[...]
```

Which makes it appear like it was installing every tool despite my configure options.

However, I do see that something has changed after `make install` because with all the `--enable-<tool>=no`, the installed `/usr/local/share/pdk` is 4MB smaller

Like you say, maybe the bigger effect would be disabling the place and route libraries that aren't helpful.  I'll try that.

itsayellow, 2025-01-27:
OK, when I look into the `/usr/local/share/pdk` directory, there are no subdirectories for the tools I disabled.  So it seems to be working.  Sorry for the bother.

I think I was spooked by the message saying it would install all the tools.  And the build process seemed the same to me (and maybe it largely is by design.)