fossi-foundation/open-pdks #178: Custom pdk conversion

msaideroglu, 2021-11-21:
Is it possible to convert a custom pdk to usable format by open-source EDA tools with open_pdks? Then, what should be the pdk's file structures?

alexhodges42, 2021-11-30:
There is some documentation here: https://openlane-docs.readthedocs.io/en/rtd/doc/PDK_STRUCTURE.html

And bundling in new PDKs is fundamentally the point of Open PDKs! You should be able to use this. I personally however had more luck using OpenROAD and just writing the couple config files that their flow requires to add a new PDK. Open PDKs and OpenLane seem like more work to integrate but should work too.

RTimothyEdwards, 2021-12-01:
@msaideroglu :  Sorry for not responding for so long.  Yes, the point of open_pdks is to try to be as flexible as possible for handling whatever files might be provided by the vendor.  I started open_pdks as a privately-developed system for installing X-Fab PDKs, and expanded it for some Global Foundries processes and a few others.  Last week I converted one of my X-Fab PDKs to work with open_pdks.

The main issue is always that since magic is the backbone of open_pdks, doing GDS and LEF reads and writes, extracting, generating netlists, etc., the first major step of converting a custom PDK is to have a techfile for magic for the process.  There is no convenient way to auto-generate one, so it can be a pretty time-consuming task.

Once you have a working tech file for magic, as long as the custom PDK provides a sufficient number of "standard formats" like GDS, LEF, liberty, verilog, and SPICE, it is relatively straightforward to create a Makefile that installs the complete PDK.

As @alexhodges42 notes, you don't necessarily need to have magic and a working tech file if the custom PDK provides all the formats you need and you don't need to generate any from scratch (or regenerate ones that are more compatible with the open source tools).  The open_pdks installer is then not doing much more than copying files around and enforcing the same filesystem structure for everything.