fossi-foundation/open-pdks #179: Generated .mag files seem to be non-deterministic?

mithro, 2021-11-22:
The output from the auto-builder seems to indicate that the .mag files generated from open_pdk are non-deterministic (IE they change even given identical input).

I'm guessing that the `.mag` file has some type of timestamp in them? I seem to recall something similar from the `skywater-pdk` generation process?

mithro, 2021-11-22:
Looks like the skywater-pdk output scripts did the following;
```python
def rewrite_out_mag(mag_in_file, mag_out_file=None):
    if mag_out_file is None:
        mag_out_file = mag_in_file

    with open(mag_in_file, 'r') as f:
        contents = f.read()

    contents = re.sub('timestamp [0-9]+', 'timestamp 0', contents)
    def basename(m):
        return 'string GDS_FILE '+os.path.basename(m.group(1))
    contents = re.sub('string GDS_FILE (.*)', basename, contents)

    with open(mag_out_file, 'w') as f:
        f.write(contents)
```

olofk, 2022-01-25:
What is the conclusion of this? Comparing mag files between the artifacts produced by open_pdks CI (e.g. https://github.com/RTimothyEdwards/open_pdks/actions/runs/1744248303) and https://foss-eda-tools.googlesource.com/skywater-pdk/output I see that the latter has timestamp=0. I also see this commit https://github.com/RTimothyEdwards/open_pdks/commit/5d450d59bb60695992f8eeaedc84cbe957077c08 Is that an acceptable middle ground?

RTimothyEdwards, 2022-01-25:
@olofk :  Yes, the commit you referenced is the best approach.  The timestamps should not be set to zero or else it gives magic fits---practically any other value would be good, but not zero, as it causes magic to force a timestamp update on everything.  The commit I made is the proper solution:  To force magic to give cells a specific timestamp so that the timestamp values can be managed but also be meaningful.  Between the recent updates to magic and to open_pdks, all the .mag files now get a timestamp that is equal to the timestamp of the last git commit of open_pdks.  As I have mentioned before, this isn't my preferred way, and I would rather have a checksum value instead of a timestamp, but I need to implement that solution.

olofk, 2022-01-25:
Agree it's much better than before since we get reproducible from a certain commit. A checksum based on file contents would be better because we still get a new timestamp on every commit even without any file changes but perhaps we have bigger fish to fry? @mithro, ok to close?

RTimothyEdwards, 2022-01-25:
@olofk :  Don't close it yet.  There is a separate issue from the timestamps where Magic enumerates instances used in a layout from some hashed database, and the enumeration is essentially random.  I don't know of any way to make the hash enumeration non-random, so the best approach is probably to just write a script that pulls the correct instance lines out of the .mag files, sorts them by name alphabetically, and then re-inserts them, all as a post-processing step.  It shouldn't take too much effort to implement.

mithro, 2022-02-01:
There seems to be something going on with the `.mag` files in the `sky130_fd_io` library see the diff at -> https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/b5d071443e656bd93ee174deb93be9f51c92d77a%5E%21/#F1 and below;
```patch
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/maglef/sky130_ef_io__gpiov2_pad.mag b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/maglef/sky130_ef_io__gpiov2_pad.mag
index 594d83f..9b3bfc5 100644
--- a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/maglef/sky130_ef_io__gpiov2_pad.mag
+++ b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/maglef/sky130_ef_io__gpiov2_pad.mag
@@ -784,7 +784,7 @@
 string LEFclass PAD INOUT
 string FIXED_BBOX 0 0 16000 39593
 string LEFview TRUE
-string GDS_FILE $PDKPATH/libs.ref/sky130_fd_io/gds/sky130_ef_io__gpiov2_pad_wrapped.gds
-string GDS_END 8250586
-string GDS_START 8232754
+string GDS_FILE $PDKPATH/libs.ref/sky130_fd_io/gds/sky130_ef_io.gds
+string GDS_END 2545096
+string GDS_START 2527264
 << end >>
```

Why is the `GDS_FILE` value changing?

commit 23c00756 (2021-11-22): Remove timestamp from .mag output files.

Fixes #179.