fossi-foundation/open-pdks #182: sky130_fd_io.gds and sky130_fd_io.lef not deterministic

mithro, 2021-11-22:
It seems like the `sky130_fd_io.gds` and `sky130_fd_io.lef` files are not deterministic? See https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/2f2e27cee9ff8d9efef4ec42ccef554706e14ae8 and builds at https://foss-eda-tools.googlesource.com/skywater-pdk/output/

RTimothyEdwards, 2021-11-22:
GDS files are timestamped. Not sure what is non-deterministic about the LEF files (most likely cause is an address-based hash table enumeration).

mithro, 2021-11-25:
The first issue seems to be the order of the cells in the `lef` file.

mithro, 2021-11-25:
```patch
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/gds/sky130_fd_io.gds b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/gds/sky130_fd_io.gds
index ee8f7fcd0..140985794 100644
Binary files a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/gds/sky130_fd_io.gds and b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/gds/sky130_fd_io.gds differ
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef
index 0984da7d2..3f56e18aa 100644
--- a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef
+++ b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef
@@ -2,12 +2,20 @@ VERSION 5.7 ;
   NOWIREEXTENSIONATPIN ON ;
   DIVIDERCHAR "/" ;
   BUSBITCHARS "[]" ;
-MACRO sky130_fd_io__overlay_vssio_hvc
+MACRO sky130_fd_io__top_ground_hvc_wpad
   CLASS PAD ;
-  FOREIGN sky130_fd_io__overlay_vssio_hvc ;
+  FOREIGN sky130_fd_io__top_ground_hvc_wpad ;
   ORIGIN 0.000 0.000 ;
   SIZE 75.000 BY 200.000 ;
   SYMMETRY X Y R90 ;
+  PIN G_PAD
```

olofk, 2022-01-25:
Is https://github.com/RTimothyEdwards/open_pdks/commit/4ea5904943f579b03fe1ed6be9427f2b04e9cc1a enough to cover the lef file determinism?

RTimothyEdwards, 2022-01-25:
It appears so.  I just did a build twice myself and got the same contents for the LEF libraries.

olofk, 2022-01-25:
That's great. Do we want another data point or can we close right away?

mithro, 2022-01-25:
Are the LEF files affected by the current non-determinism in the `.mag` files? #179