fossi-foundation/open-pdks #183: Output PDK has two magic core dumps in it?

mithro, 2021-11-22:
See https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/refs/heads/main/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/core
```
tansell@tansell-glaptop:~/gob/foss-eda-tools/skywater-pdk-output$ find -name core
./usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/mag/core
./usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/core
tansell@tansell-glaptop:~/gob/foss-eda-tools/skywater-pdk-output$ file ./usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/mag/core
./usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/mag/core: ELF 64-bit LSB core file, x86-64, version 1 (SYSV), SVR4-style, from '/host/out/magic/lib/magic/tcl/magicdnull -nowrapper -dnull -noconsole', real uid: 0, effective uid: 0, real gid: 0, effective gid: 0, execfn: '/host/out/magic/lib/magic/tcl/magicdnull', platform: 'x86_64'
tansell@tansell-glaptop:~/gob/foss-eda-tools/skywater-pdk-output$ file ./usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/core
./usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/core: ELF 64-bit LSB core file, x86-64, version 1 (SYSV), SVR4-style, from '/host/out/magic/lib/magic/tcl/magicdnull -nowrapper -dnull -noconsole', real uid: 0, effective uid: 0, real gid: 0, effective gid: 0, execfn: '/host/out/magic/lib/magic/tcl/magicdnull', platform: 'x86_64'
```

RTimothyEdwards, 2021-11-22:
Can't fix it unless you tell me what it was doing when it crashed.

mithro, 2021-11-22:
First step would be figuring out why Magic segfaulting didn't stop the install script?

mithro, 2021-11-23:
```
LEF read: Processed 57 lines.
Writing all magic database files
Writing 'sky130_fd_sc_hd__decap_12'
Writing 'sky130_ef_sc_hd__fill_4'
Writing 'sky130_fd_sc_hd__decap_6'
Error message output from magic script:
Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "sky130_fd_sc_hd__fill_4" is used but not defined in this file.
Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "sky130_fd_sc_hd__fill_2" is used but not defined in this file.
Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): Cell "sky130_fd_sc_hd__decap_6" is used but not defined in this file.
Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__decap_6 was used but not defined.
Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__fill_2 was used but not defined.
Error while reading cell "sky130_ef_sc_hd__fill_12" (byte position 372): cell sky130_fd_sc_hd__fill_4 was used but not defined.
File sky130_fd_sc_hd__decap_6 contained format error
"sky130_fd_sc_hd__decap_6" has a zero timestamp; it should be written out
    to establish a correct timestamp.
freeMagic called with NULL argument.
Traceback (most recent call last):
  File "../common/foundry_install.py", line 1648, in <module>
    cwd = destlibdir,
  File "../common/foundry_install.py", line 256, in subprocess_run
    raise SystemError(emsg)
SystemError: Command magic failed with exit code: -11
  magic -dnull -noconsole
make[3]: *** [Makefile:1048: digital-hd-a] Error 1
make[2]: *** [Makefile:856: vendor-a] Error 2
make[1]: *** [Makefile:583: all-a] Error 2
make[3]: Leaving directory '/home/runner/work/open_pdks/open_pdks/sky130'
make[2]: Leaving directory '/home/runner/work/open_pdks/open_pdks/sky130'
make[1]: Leaving directory '/home/runner/work/open_pdks/open_pdks/sky130'
make: *** [Makefile:58: build-pdk] Error 2
Error: Process completed with exit code 2.
```

mithro, 2021-11-23:
Trying to track down the magic segfault @ https://github.com/RTimothyEdwards/magic/issues/102

RTimothyEdwards, 2021-11-23:
I should be able to work with that.

mithro, 2021-11-23:
With #189 we seems to consistently get a segfault on the `sky130_fd_sc_hd` library, as shown at https://github.com/RTimothyEdwards/open_pdks/runs/4304630430?check_suite_focus=true

I have yet to reproduce the segfault on my local machine....

RTimothyEdwards, 2021-11-25:
Should be fixed in magic 8.3.232.

closed 2021-12-01 by RTimothyEdwards