fossi-foundation/open-pdks #184: Fill .spice files seem to be non-deterministic

mithro, 2021-11-22:
```
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_ef_sc_hd__fill_8.spice b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_ef_sc_hd__fill_8.spice
index 6d2d4412..4d213b37 100644
--- a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_ef_sc_hd__fill_8.spice
+++ b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_ef_sc_hd__fill_8.spice
@@ -1,7 +1,7 @@
 * NGSPICE file created from sky130_ef_sc_hd__fill_8.ext - technology: sky130A
 
 .subckt sky130_ef_sc_hd__fill_8 VGND VPWR VPB VNB
-C0 VPWR a_31_305# 0.14fF
-C1 a_31_39# VGND 0.14fF
+C0 a_31_305# VPWR 0.14fF
+C1 VGND a_31_39# 0.14fF
 .ends
 
```

mithro, 2021-11-22:
Seems to affect `sky130_ef_sc_hd__fill_8.spice` and `sky130_ef_sc_hvl__fill_8.spice`

RTimothyEdwards, 2021-11-22:
Why is extraction being done with parasitic capacitances?

mithro, 2021-11-22:
I don't know, you wrote the scripts which create these files?

RTimothyEdwards, 2021-12-01:
I just added an option to distinguish between generating SPICE files with or without parasitics, so this should not be a problem in the future.

olofk, 2022-01-25:
Is there any action to take here or simply a user option to generate with/without parasitics?

RTimothyEdwards, 2022-01-25:
@olofk :  open_pdks is generating these SPICE netlists without parasitics now, so there should no longer be any difference between these files from run to run of the install.

olofk, 2022-01-25:
A candidate for closing then?

mithro, 2022-01-25:
So while the parasitics have been fixed, the order of things in the files seems to still change.

olofk, 2022-01-25:
@mithro Order of things like in the diff you show above? I.e. `a_31_305#` and `VPWR` being swapped? (I have no idea what spice files looks like so I might have misunderstood)

mithro, 2022-01-25:
@olofk - Correct.

mithro, 2022-01-26:
Spice files still seem to change randomly, see diffs @ https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/6cf361415849f0d802413e2f419abd5d29ff4d09

mithro, 2022-01-26:
The delta in the spice files seems very weird;
```
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_pr/spice/sky130_fd_pr__esd_nfet_05v0_nvt.pm3.spice b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_pr/spice/sky130_fd_pr__esd_nfet_05v0_nvt.pm3.spice
index e2deb23..c6869e0 100644
--- a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_pr/spice/sky130_fd_pr__esd_nfet_05v0_nvt.pm3.spice
+++ b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_pr/spice/sky130_fd_pr__esd_nfet_05v0_nvt.pm3.spice
@@ -13,27 +13,27 @@
 * limitations under the License.
 
 * SKY130 Spice File.
-.param sky130_fd_pr__nfet_g5v0d10v5__toxe_slope_spectre = 0.0
-.param sky130_fd_pr__nfet_g5v0d10v5__vth0_slope_spectre = 0.0
-.param sky130_fd_pr__nfet_g5v0d10v5__voff_slope_spectre = 0.0
-.param sky130_fd_pr__nfet_g5v0d10v5__nfactor_slope_spectre = 0.0
+.param sky130_fd_pr__esd_nfet_05v0_nvt__toxe_slope_spectre = 0.0
+.param sky130_fd_pr__esd_nfet_05v0_nvt__vth0_slope_spectre = 0.0
+.param sky130_fd_pr__esd_nfet_05v0_nvt__voff_slope_spectre = 0.0
+.param sky130_fd_pr__esd_nfet_05v0_nvt__nfactor_slope_spectre = 0.0
```