fossi-foundation/open-pdks #188: (UNNAMED).spice?

mithro, 2021-11-23:
What is up with this `(UNNAMED).spice` file in the output?
```
Nothing was replaced in 'sky130A/libs.ref/sky130_fd_pr/spice/(UNNAMED).spice'.
```

RTimothyEdwards, 2021-11-23:
(UNNAMED) is the default name of a cell in magic if you don't save it under a different name.  I wrote some code that's supposed to delete the cell if it's empty and something else has been loaded (e.g., from reading a GDS file), but apparently there are multiple ways to bypass that code.  This could happen, perhaps, if a GDS file had a different name from a cell, such that the cell is not loaded automatically after reading GDS.  I could add simple behavior to cause magic to refuse to run "extract" on "(UNNAMED)" and insist that it should have a real name first.

RTimothyEdwards, 2021-11-25:
Should be fixed in magic version 8.3.232.

closed 2021-12-01 by RTimothyEdwards