fossi-foundation/open-pdks #190: Add the io library to the CI build

mithro, 2021-11-23:
It seems like the IO library is not being built in CI?

https://github.com/RTimothyEdwards/open_pdks/blob/5f6daa1abdc7acd274c5484682496ba16fe182e3/.github/workflows/ci.yml#L15-L21

mithro, 2021-11-24:
Just trying to add `sky130_fd_io` to the library list above gets you the following error;
```
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk && \
	git submodule update --init libraries/sky130_fd_io/latest && \
	make -j2 sky130_fd_io
Submodule 'libraries/sky130_fd_io/latest' (https://github.com/google/skywater-pdk-libs-sky130_fd_io.git) registered for path 'libraries/sky130_fd_io/latest'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/libraries/sky130_fd_io/latest'...
From https://github.com/google/skywater-pdk-libs-sky130_fd_io
 * branch            01b18699b4102d8e54ad1406b3991ecb652e5aee -> FETCH_HEAD
Submodule 'third_party/make-env' (https://github.com/SymbiFlow/make-env.git) registered for path 'third_party/make-env'
Submodule path 'libraries/sky130_fd_io/latest': checked out '01b18699b4102d8e54ad1406b3991ecb652e5aee'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/third_party/make-env'...
make[1]: Entering directory '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk; git submodule update --init third_party/make-env
Submodule path 'third_party/make-env': checked out '9b07ad2bb62fbf8af789c9e4669715c974b4912d'
make[1]: *** No rule to make target 'sky130_fd_io'.  Stop.
make[1]: Leaving directory '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'
make: *** [Makefile:40: skywater-library] Error 2
git clone https://github.com/google/skywater-pdk.git /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'...
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk && \
	git checkout -qf b5ff6174d9aa921f670f12a2893969fe4a516f6c
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk && \
	git submodule update --init libraries/sky130_fd_io/latest && \
	make -j2 sky130_fd_io
Submodule 'libraries/sky130_fd_io/latest' (https://github.com/google/skywater-pdk-libs-sky130_fd_io.git) registered for path 'libraries/sky130_fd_io/latest'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/libraries/sky130_fd_io/latest'...
From https://github.com/google/skywater-pdk-libs-sky130_fd_io
 * branch            01b18699b4102d8e54ad1406b3991ecb652e5aee -> FETCH_HEAD
Submodule path 'libraries/sky130_fd_io/latest': checked out '01b18699b4102d8e54ad1406b3991ecb652e5aee'
make[1]: Entering directory '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk; git submodule update --init third_party/make-env
Submodule 'third_party/make-env' (https://github.com/SymbiFlow/make-env.git) registered for path 'third_party/make-env'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/third_party/make-env'...
Submodule path 'third_party/make-env': checked out '9b07ad2bb62fbf8af789c9e4669715c974b4912d'
make[1]: *** No rule to make target 'sky130_fd_io'.  Stop.
make[1]: Leaving directory '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'
make: *** [Makefile:40: skywater-library] Error 2
git clone https://github.com/google/skywater-pdk.git /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'...
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk && \
	git checkout -qf b5ff6174d9aa921f670f12a2893969fe4a516f6c
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk && \
	git submodule update --init libraries/sky130_fd_io/latest && \
	make -j2 sky130_fd_io
Submodule 'libraries/sky130_fd_io/latest' (https://github.com/google/skywater-pdk-libs-sky130_fd_io.git) registered for path 'libraries/sky130_fd_io/latest'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/libraries/sky130_fd_io/latest'...
From https://github.com/google/skywater-pdk-libs-sky130_fd_io
 * branch            01b18699b4102d8e54ad1406b3991ecb652e5aee -> FETCH_HEAD
Submodule path 'libraries/sky130_fd_io/latest': checked out '01b18699b4102d8e54ad1406b3991ecb652e5aee'
make[1]: Entering directory '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk; git submodule update --init third_party/make-env
Submodule 'third_party/make-env' (https://github.com/SymbiFlow/make-env.git) registered for path 'third_party/make-env'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/third_party/make-env'...
Submodule path 'third_party/make-env': checked out '9b07ad2bb62fbf8af789c9e4669715c974b4912d'
make[1]: *** No rule to make target 'sky130_fd_io'.  Stop.
make[1]: Leaving directory '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'
make: *** [Makefile:40: skywater-library] Error 2
git clone https://github.com/google/skywater-pdk.git /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'...
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk && \
	git checkout -qf b5ff6174d9aa921f670f12a2893969fe4a516f6c
cd /home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk && \
	git submodule update --init libraries/sky130_fd_io/latest && \
	make -j2 sky130_fd_io
Submodule 'libraries/sky130_fd_io/latest' (https://github.com/google/skywater-pdk-libs-sky130_fd_io.git) registered for path 'libraries/sky130_fd_io/latest'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/libraries/sky130_fd_io/latest'...
[... cut ...]
Submodule 'third_party/make-env' (https://github.com/SymbiFlow/make-env.git) registered for path 'third_party/make-env'
Cloning into '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk/third_party/make-env'...
Submodule path 'third_party/make-env': checked out '9b07ad2bb62fbf8af789c9e4669715c974b4912d'
make[1]: *** No rule to make target 'sky130_fd_io'.  Stop.
make[1]: Leaving directory '/home/runner/work/open_pdks/open_pdks/pdks/skywater-pdk'
make: *** [Makefile:40: skywater-library] Error 2
Error: Process completed with exit code 2.
```

olofk, 2022-01-25:
I have not had issues adding the sky130_fd_io library in my experiments. Is this still an issue?

RTimothyEdwards, 2022-01-25:
It shouldn't have been an issue in the first place.  I've been using the `sky130_fd_io` library from skywater-pdk.  But I think this has more to do with the fact that the CI is running `make` on the `sky130_fd_io` directory.  I'm not sure what the CI is doing.  Apparently it is running its own installation of skywater-pdk instead of letting open_pdks take care of it.  Open_pdks only does a `git submodule init`, `git submodule update`, and then (from the repository top level) `make timing`.