fossi-foundation/open-pdks #193: The "all" build takes almost 2 hours due to capturing

mithro, 2021-11-25:
From https://github.com/RTimothyEdwards/open_pdks/runs/4327353828?check_suite_focus=true 

The run `succeeded 2 hours ago in 1h 47m 35s` of which `1h 16m 53s` was used to create the tarball while only `29m 12s` was used in the building.

This is significantly longer than the other builds which average ~30 minutes;
[image]

RTimothyEdwards, 2021-11-26:
This is from the CI or something?  I don't have any idea what it's doing.

mithro, 2021-11-26:
Yes, this is the GitHub Actions CI.

It is probably caused by using `--xz` compression on this line -> https://github.com/RTimothyEdwards/open_pdks/blob/master/.github/capture.sh#L43

RTimothyEdwards, 2021-11-26:
I have no idea what that script is doing, or what it is intended to do.

mithro, 2021-11-26:
The captures the output from the CI run so that GitHub Actions uploads it as described in https://github.com/RTimothyEdwards/open_pdks/pull/192

RTimothyEdwards, 2021-11-26:
It is then probably more urgent to create a "debug level" option to open_pdks and restrict the output to have much less output by default.

mithro, 2021-11-27:
Testing using gzip compression rather than xz @ https://github.com/mithro/open_pdks/commit/eb09f468da3e4f4514426267beb9f8a0f936c4e3 and https://github.com/mithro/open_pdks/actions/runs/1511101558

We will see if it makes much of a difference...

mithro, 2021-11-29:
Fixed by #196.

commit eb09f468 (2021-11-30): Use much faster gzip compression.

Potentially fixes #193.
closed 2021-11-29 by mithro