fossi-foundation/open-pdks #195: Non-deterministic output of script

mithro, 2021-11-28:
The output of `open_pdks` is non-deterministic -- IE the exact same input does not produce the exact same output.

This is demonstrated by the PDK being built every 30 minutes and published in the repository at https://foss-eda-tools.googlesource.com/skywater-pdk/output.git

Once I zeroed out the timestamps and gds date stamps (script found at https://foss-eda-tools.googlesource.com/skywater-pdk/builder/+/refs/heads/main/open_pdks/post-open_pdks.sh), the root cause seems to be the `.lef` files?

Sometimes the diff is small;
```
tansell@tansell-glapstation:~/gob/foss-eda-tools/skywater-pdk-output$ git diff HEAD~1 --stat                                                                                                                         
 build.info                                                                                    |      4 +-                                                                                                          
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef                        | 106524 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++----------------------------------------------------------------------------------------------------------------------------------------------------------------
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__simple_pad_and_busses.mag |      3 +-                                                                                                          
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef                  |  72360 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-------------------------------------------------------------------------------------------------------------
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_ef_sc_hd__fill_8.spice      |      4 +-                                                                                                          
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hvl/lef/sky130_fd_sc_hvl.lef                |  18246 +++++++++++++++++++++++++++----------------------------
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hvl/maglef/unithvdbl.mag                    |      9 +                                                                                                           
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_sc_hvl/spice/sky130_ef_sc_hvl__fill_8.spice    |      6 +-                                                                                                          
 8 files changed, 98586 insertions(+), 98570 deletions(-)      
```

Sometimes the diff is large;
```
tansell@tansell-glapstation:~/gob/foss-eda-tools/skywater-pdk-output$ git diff HEAD~2 --stat                                                                                                                         
 build.info                                                                                                              |      4 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef                                                  | 106948 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++---------------------------------------------------------------------------------------------------------------------------------------------------
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_ef_io__gpiov2_pad.mag                                      |      8 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__amux_switch_1v2b.mag                                |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__amx_inv1.mag                                        |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_bus_slice_m4.mag                                |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_cclat.mag                                       |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_hldv2.mag                                   |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_ls.mag                                      |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_ls_1v2.mag                                  |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_ls_en_1_v2.mag                              |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_ls_octl.mag                                 |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_ls_ovt.mag                                  |    346 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_ls_ovtv2.mag                                |    346 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_ls_v2.mag                                   |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_ctl_lsv2.mag                                    |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_inv_x1_dnwv2.mag                                |     22 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_inv_x1_dnwv2_1.mag                              |     12 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_nand2_dnwv2.mag                                 |     38 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_nor2_dnwv2.mag                                  |     38 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_nor2_dnwv2_1.mag                                |     30 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_opath_datoev2.mag                               |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pdpredrvr_pbiasv2.mag                           |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pdpredrvr_strong_slow.mag                       |     76 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pdpredrvr_strong_slowv2.mag                     |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pdpredrvr_weak.mag                              |     74 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pdpredrvr_weakv2.mag                            |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pudrvr_strong_slowv2.mag                        |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pudrvr_weakv2.mag                               |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pupredrvr_strong_slow.mag                       |     90 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_pupredrvr_strong_slowv2.mag                     |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_res_weak.mag                                    |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_res_weak_bentbigres.mag                         |      6 +-
 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__com_res_weak_v2.mag                                 |    120 +-
```

The related GitHub issues are;

 - [ ] #182 
 - [ ] #184

mithro, 2021-11-28:
The issue with the `.lef` output seems to be related to not sorting the output?

mithro, 2021-11-28:
```patch
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef
index 512003217..7acf0f5f6 100644
--- a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef
+++ b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef
@@ -2,51225 +2,51128 @@ VERSION 5.7 ;
   NOWIREEXTENSIONATPIN ON ;
   DIVIDERCHAR "/" ;
   BUSBITCHARS "[]" ;
-MACRO sky130_fd_io__overlay_vssa_hvc
+MACRO sky130_fd_io__overlay_vssa_lvc
   CLASS PAD ;
-  FOREIGN sky130_fd_io__overlay_vssa_hvc ;
+  FOREIGN sky130_fd_io__overlay_vssa_lvc ;
   ORIGIN 0.000 0.000 ;
-  SIZE 75.000 BY 200.000 ;
+  SIZE 75.000 BY 198.000 ;
   SYMMETRY X Y R90 ;
-  PIN VDDA
+  PIN VDDIO_Q
     DIRECTION INOUT ;
     USE POWER ;
     PORT
       LAYER met5 ;
-        RECT 0.000 15.035 0.965 18.285 ;
+        RECT 0.000 62.185 1.270 66.435 ;
     END
     PORT
       LAYER met5 ;
-        RECT 74.035 15.035 75.000 18.285 ;
+        RECT 73.730 62.185 75.000 66.435 ;
     END
     PORT
       LAYER met4 ;
-        RECT 74.035 14.935 75.000 18.385 ;
+        RECT 73.730 62.085 75.000 66.535 ;
     END
     PORT
       LAYER met4 ;
-        RECT 0.000 14.935 0.965 18.385 ;
+        RECT 0.000 62.085 1.270 66.535 ;
     END
-  END VDDA
-  PIN VDDIO
+  END VDDIO_Q
+  PIN VSSA
     DIRECTION INOUT ;
-    USE POWER ;
+    USE GROUND ;
     PORT
       LAYER met5 ;
-        RECT 73.730 70.035 75.000 94.985 ;
+        RECT 73.730 45.735 75.000 54.735 ;
     END
     PORT
       LAYER met5 ;
-        RECT 73.730 19.885 75.000 24.335 ;
+        RECT 73.730 34.840 75.000 38.085 ;
```

mithro, 2021-11-28:
The output in the `.mag` files looks like;
```diff
tansell@tansell-glapstation:~/gob/foss-eda-tools/skywater-pdk-output$ git diff HEAD~1 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__simple_pad_and_busses.mag
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__simple_pad_and_busses.mag b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__simple_pad_and_busses.mag
index 401415f22..3cdecbe0a 100644
--- a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__simple_pad_and_busses.mag
+++ b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__simple_pad_and_busses.mag
@@ -3,7 +3,8 @@ tech sky130A
 magscale 1 2
 timestamp 0
 << checkpaint >>
-rect -1252 -1259 16268 41261
+rect -1252 -852 16268 41261
+rect -1252 -1259 16118 -852
 << dnwell >>
 rect 670 9738 14334 36138
 << nwell >>
```

```patch
tansell@tansell-glapstation:~/gob/foss-eda-tools/skywater-pdk-output$ git diff HEAD~2 usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__gpio_ovtv2_amux_ctl_ls_i2c_fix.mag
diff --git a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__gpio_ovtv2_amux_ctl_ls_i2c_fix.mag b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__gpio_ovtv2_amux_ctl_ls_i2c_fix.mag
index 8a68727da..b0980b86e 100644
--- a/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__gpio_ovtv2_amux_ctl_ls_i2c_fix.mag
+++ b/usr/local/share/pdk/sky130A/libs.ref/sky130_fd_io/mag/sky130_fd_io__gpio_ovtv2_amux_ctl_ls_i2c_fix.mag
@@ -1942,54 +1942,54 @@ rect -1213 -282 -1208 -226
 rect -1152 -282 -1101 -226
 rect -1045 -282 -1040 -226
 rect -1213 -287 -1040 -282
-use sky130_fd_pr__nfet_01v8__example_55959141808468  sky130_fd_pr__nfet_01v8__example_55959141808468_0
+use sky130_fd_pr__pfet_01v8__example_55959141808458  sky130_fd_pr__pfet_01v8__example_55959141808458_0
 timestamp 0
-transform -1 0 2161 0 1 758
-box -28 0 916 97
-use sky130_fd_pr__nfet_01v8__example_55959141808467  sky130_fd_pr__nfet_01v8__example_55959141808467_0
+transform 0 1 206 -1 0 1259
+box -28 0 284 63
+use sky130_fd_pr__nfet_01v8__example_55959141808461  sky130_fd_pr__nfet_01v8__example_55959141808461_0
 timestamp 0
-transform -1 0 894 0 -1 748
-box -28 0 316 97
-use sky130_fd_pr__nfet_01v8__example_55959141808466  sky130_fd_pr__nfet_01v8__example_55959141808466_0
+transform 1 0 1496 0 -1 363
+box -28 0 440 131
+use sky130_fd_pr__nfet_01v8__example_55959141808463  sky130_fd_pr__nfet_01v8__example_55959141808463_0
 timestamp 0
-transform -1 0 550 0 -1 748
-box -28 0 316 97
+transform 1 0 736 0 1 114
+box -28 0 596 97
 use sky130_fd_pr__nfet_01v8__example_55959141808465  sky130_fd_pr__nfet_01v8__example_55959141808465_0
 timestamp 0
 transform 1 0 2089 0 -1 363
 box -28 0 128 97
-use sky130_fd_pr__nfet_01v8__example_55959141808464  sky130_fd_pr__nfet_01v8__example_55959141808464_0
-timestamp 0
-transform -1 0 2161 0 -1 1288
-box -28 0 916 97
-use sky130_fd_pr__nfet_01v8__example_55959141808463  sky130_fd_pr__nfet_01v8__example_55959141808463_0
+use sky130_fd_pr__nfet_01v8__example_55959141808466  sky130_fd_pr__nfet_01v8__example_55959141808466_0
 timestamp 0
-transform 1 0 736 0 1 114
-box -28 0 596 97
-use sky130_fd_pr__nfet_01v8__example_55959141808461  sky130_fd_pr__nfet_01v8__example_55959141808461_0
+transform -1 0 550 0 -1 748
+box -28 0 316 97
+use sky130_fd_pr__nfet_01v8__example_55959141808467  sky130_fd_pr__nfet_01v8__example_55959141808467_0
 timestamp 0
-transform 1 0 1496 0 -1 363
-box -28 0 440 131
-use sky130_fd_pr__pfet_01v8__example_55959141808460  sky130_fd_pr__pfet_01v8__example_55959141808460_0
+transform -1 0 894 0 -1 748
+box -28 0 316 97
+use sky130_fd_pr__nfet_01v8__example_55959141808468  sky130_fd_pr__nfet_01v8__example_55959141808468_0
 timestamp 0
-transform 0 1 817 1 0 1159
-box -28 0 128 63
-use sky130_fd_pr__pfet_01v8__example_55959141808460  sky130_fd_pr__pfet_01v8__example_55959141808460_1
+transform -1 0 2161 0 1 758
+box -28 0 916 97
+use sky130_fd_pr__pfet_01v8__example_55959141808460  sky130_fd_pr__pfet_01v8__example_55959141808460_3
 timestamp 0
-transform 0 1 817 1 0 1003
+transform 0 1 544 1 0 1003
 box -28 0 128 63
 use sky130_fd_pr__pfet_01v8__example_55959141808460  sky130_fd_pr__pfet_01v8__example_55959141808460_2
 timestamp 0
 transform 0 1 544 1 0 1159
 box -28 0 128 63
-use sky130_fd_pr__pfet_01v8__example_55959141808460  sky130_fd_pr__pfet_01v8__example_55959141808460_3
+use sky130_fd_pr__pfet_01v8__example_55959141808460  sky130_fd_pr__pfet_01v8__example_55959141808460_1
 timestamp 0
-transform 0 1 544 1 0 1003
+transform 0 1 817 1 0 1003
 box -28 0 128 63
-use sky130_fd_pr__pfet_01v8__example_55959141808458  sky130_fd_pr__pfet_01v8__example_55959141808458_0
+use sky130_fd_pr__pfet_01v8__example_55959141808460  sky130_fd_pr__pfet_01v8__example_55959141808460_0
 timestamp 0
-transform 0 1 206 -1 0 1259
-box -28 0 284 63
+transform 0 1 817 1 0 1159
+box -28 0 128 63
+use sky130_fd_pr__nfet_01v8__example_55959141808464  sky130_fd_pr__nfet_01v8__example_55959141808464_0
+timestamp 0
+transform -1 0 2161 0 -1 1288
+box -28 0 916 97
 << labels >>
 flabel comment s 741 1066 741 1066 0 FreeSans 600 0 0 0 FBK_N
 flabel metal1 s 771 472 799 500 3 FreeSans 280 0 0 0 IN_B
```

RTimothyEdwards, 2021-12-01:
Magic does not sort uses in its .mag output;  they are an enumerated hash table.  Forcing enumeration in alphabetical order would seriously undermine the efficiency of writing output files.  However, given that all the files in the PDK are small, not entire chips, then maybe I could implement a command-line option to force alphabetical ordering.

The "checkpaint" geometry is enumerating areas that need DRC checking;  they shouldn't be in the library files at all.  There is probably some way to automatically keep them out of the output.  I'll see if I can figure out what.

The order of macros in the LEF library is supposed to be sorted by open_pdks, so I will need to look into why that is apparently not happening.

RTimothyEdwards, 2021-12-01:
I think your snippet with the "checkpaint" diff just solved a major mystery that has been bothering me for a long time.  The "checkpaint" exists because a statement to force cells read from GDS not to get DRC checks was missing in the foundry_install script.  That "checkpaint" causes every cell having it to be checked as soon as it's loaded into magic, and then magic wants to update all the timestamps and write out "clean" versions of all those cells (which it can't ever do, because they're in a read-only directory).  Fixing that will make layout editing run a lot more smoothly.  Won't solve any of your other problems, though.

FYI, at least for me, LEF macros are sorted alphabetically in the generated LEF library, and there are no differences when I run the install multiple times.

mithro, 2021-12-01:
I'm still seeing non-determinism with the latest versions;
```
     SKY130A PDK version: v9.3.484-363
    
     Which was built using:
           Builder: v0.0-34-gf6499e7
      SkyWater PDK: v0.0.0-329-gb5ff617
         Open-PDKs: v1.0.250-0-g9638955
             Magic: v8.3.234-0-g6574713
```

See https://foss-eda-tools.googlesource.com/skywater-pdk/output/+log/refs/heads/main

mithro, 2021-12-08:
We are still seeing non-determinism with the latest version as of today;
```
 SKY130A PDK version: v9.3.489-363

 Which was built using:
       Builder: v0.0-34-gf6499e7
  SkyWater PDK: v0.0.0-329-gb5ff617
     Open-PDKs: v1.0.251-0-g89f6ff4
         Magic: v8.3.238-0-g9f70113
```
See https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/a7ac7db3a51b475e761c1d6b539a11d0519ee982 and https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/bdc557c98264ee62ea2b4a6fd96ddec6444a8925 for example.

mithro, 2021-12-08:
I'm still seeing non-determinism with LEF output which seems to be due to both cell order. There is potentially pins inside cell ordering issue, but it is hard to tell with the cell order issue.

mithro, 2021-12-08:
Actually, the latest version is;
```
 SKY130A PDK version: v9.3.491-363

 Which was built using:
       Builder: v0.0-34-gf6499e7
  SkyWater PDK: v0.0.0-329-gb5ff617
     Open-PDKs: v1.0.252-0-gb8c95cb
         Magic: v8.3.239-0-g552d6de
```

mithro, 2021-12-08:
With that version I confirm that I'm still seeing the LEF output ordering issue.

See https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/34c884c6c28b5c68edf0694c0bdf53f1b157edab%5E%21/ and https://foss-eda-tools.googlesource.com/skywater-pdk/output/+/1f1f075f76acdaeec339e95132c6909e7a2d7b07%5E%21/ for example.

mithro, 2021-12-08:
In case it matters, the script which actually runs the PDK generation can be found at https://foss-eda-tools.googlesource.com/skywater-pdk/builder/+/refs/heads/main/open_pdks/build-open_pdks.sh#82

mithro, 2021-12-08:
Magic was also built with the following script -> https://foss-eda-tools.googlesource.com/skywater-pdk/builder/+/refs/heads/main/open_pdks/build-magic.sh

mithro, 2021-12-08:
FYI The command used with `open_pdks` installer was;
```
 ./configure --enable-sky130-pdk=/host/github/google/skywater-pdk --enable-alpha-sky130 --prefix=/usr/local --enable-klayout --enable-magic --enable-netgen --enable-irsim --enable-openlane --enable-qflow --enable-xschem
```
It was built with;
```
make -j4
```
It was installed with;
```
make DESTDIR=/host/out/pdk-all install
```