fossi-foundation/open-pdks #201: Monte Carlo example

lhrodovalho, 2021-12-19:
Is there any example of any monte carlo simulation using the sky130A ngspice libs? The only example of monte carlo simulation I've seen for sky130A was the one which comes with the xschem examples. I've tried to use sky130.lib.spice mc and mc_mm_switch=1 and it doesn't seem to work.

RTimothyEdwards, 2021-12-19:
You should not need to set `mc_mm_switch`, but using `.lib <path_to>/sky130.lib.spice mc` should work.  What are you observing that leads you to conclude that this doesn't work?  Don't forget to use `.option SEED=random` or else the "random variation" will be exactly repeatable (see the ngspice manual section "Analyses and Output Contro -> Simulator Variables -> General Options" and the section "Statistical circuit analysis -> Monte-Carlo Simulation").

lhrodovalho, 2021-12-19:
It is working now. 

Here is the code I'm using. It's just a cascode current mirror

`* cascode mirror monte carlo testbench

* Include SkyWater sky130 device models
.lib "/usr/share/pdk/sky130A/libs.tech/ngspice/sky130.lib.spice" mc
.param mc_mm_switch=1

.subckt n1_1 D G S B
X0 D G S B sky130_fd_pr__nfet_01v8 ad=4e+11p pd=2.8e+06u as=4e+11p ps=2.8e+06u w=1.0 l=0.5 m=1
.ends

.subckt mirror ii io vssa
x1b ii ii n1   vssa n1_1
x1a n1 n1 vssa vssa n1_1

x2b io ii n2   vssa n1_1
x2a n2 n1 vssa vssa n1_1
.ends

* supply voltages
vdda	vdda 0 1.8
vssa	vssa 0 0.0

* input signals
ii vdda ii 100n

* DUT
xdut ii io vssa mirror
vo io vssa 0.9

.option gmin=1e-15
.option seed=random
.control

  set appendwrite
	      
  let run = 1
  dowhile run <= 100
      reset
      op
      print -i(vo)
      wrdata cmirror.txt -i(vo)
      let run = run + 1
  end
    
.endc

.end`

The results are a little weird. It's repeating most of results.

` 0.00000000e+00  1.25723549e-07 
 0.00000000e+00  1.19438386e-07 
 0.00000000e+00  1.19438386e-07 
 0.00000000e+00  1.19438386e-07 
 0.00000000e+00  7.36299401e-08 
 0.00000000e+00  7.36299401e-08 
 0.00000000e+00  7.36299401e-08 
 0.00000000e+00  6.57856318e-08 
 0.00000000e+00  6.57856318e-08 
 0.00000000e+00  6.57856318e-08 
 0.00000000e+00  7.60332873e-08 
 0.00000000e+00  7.60332873e-08 
 0.00000000e+00  7.60332873e-08 
 0.00000000e+00  8.92142086e-08 
 0.00000000e+00  8.92142086e-08 
 0.00000000e+00  8.92142086e-08 
 0.00000000e+00  7.67733530e-08 
 0.00000000e+00  7.67733530e-08 
 0.00000000e+00  7.67733530e-08 
 0.00000000e+00  7.67733530e-08 
 0.00000000e+00  1.02408988e-07 
 0.00000000e+00  1.02408988e-07 
 0.00000000e+00  1.02408988e-07 
 0.00000000e+00  1.40886246e-07 
 0.00000000e+00  1.40886246e-07 
 0.00000000e+00  1.40886246e-07 
 0.00000000e+00  1.16252589e-07 
 0.00000000e+00  1.16252589e-07 
 0.00000000e+00  1.16252589e-07 
 0.00000000e+00  1.28616915e-07 
 0.00000000e+00  1.28616915e-07 
 0.00000000e+00  1.28616915e-07 
 0.00000000e+00  7.74502629e-08 
 0.00000000e+00  7.74502629e-08 
 0.00000000e+00  7.74502629e-08 
 0.00000000e+00  1.79339098e-07 
 0.00000000e+00  6.65927789e-08 
 0.00000000e+00  6.65927789e-08 
 0.00000000e+00  1.09944515e-07 
 0.00000000e+00  1.09944515e-07 
 0.00000000e+00  1.09944515e-07 
 0.00000000e+00  6.03082056e-08 
 0.00000000e+00  6.03082056e-08 
 0.00000000e+00  6.03082056e-08 
 0.00000000e+00  7.45341353e-08 
 0.00000000e+00  7.45341353e-08 
 0.00000000e+00  1.25462960e-07 
 0.00000000e+00  1.25462960e-07 
 0.00000000e+00  5.40797241e-08 
 0.00000000e+00  5.40797241e-08 
 0.00000000e+00  1.19822149e-07 
 0.00000000e+00  1.19822149e-07 
 0.00000000e+00  9.66296963e-08 
 0.00000000e+00  9.66296963e-08 
 0.00000000e+00  9.66296963e-08 
 0.00000000e+00  1.09885928e-07 
 0.00000000e+00  1.09885928e-07 
 0.00000000e+00  1.38002183e-07 
 0.00000000e+00  1.38002183e-07 
 0.00000000e+00  1.38002183e-07 
 0.00000000e+00  9.27004353e-08 
 0.00000000e+00  9.27004353e-08 
 0.00000000e+00  9.27004353e-08 
 0.00000000e+00  1.55837423e-07 
 0.00000000e+00  1.55837423e-07 
 0.00000000e+00  1.34809293e-07 
 0.00000000e+00  1.34809293e-07 
 0.00000000e+00  1.34809293e-07 
 0.00000000e+00  6.48953742e-08 
 0.00000000e+00  6.48953742e-08 
 0.00000000e+00  8.71935954e-08 
 0.00000000e+00  8.71935954e-08 
 0.00000000e+00  8.71935954e-08 
 0.00000000e+00  8.79232191e-08 
 0.00000000e+00  8.79232191e-08 
 0.00000000e+00  8.79232191e-08 
 0.00000000e+00  1.76363110e-07 
 0.00000000e+00  1.76363110e-07 
 0.00000000e+00  9.02236918e-08 
 0.00000000e+00  9.02236918e-08 
 0.00000000e+00  9.02236918e-08 
 0.00000000e+00  1.64945923e-07 
 0.00000000e+00  1.64945923e-07 
 0.00000000e+00  1.64945923e-07 
 0.00000000e+00  7.64482972e-08 
 0.00000000e+00  6.60547929e-08 
 0.00000000e+00  6.60547929e-08 
 0.00000000e+00  1.42333200e-07 
 0.00000000e+00  1.42333200e-07 
 0.00000000e+00  6.29370159e-08 
 0.00000000e+00  6.29370159e-08 
 0.00000000e+00  1.30169292e-07 
 0.00000000e+00  1.30169292e-07 
 0.00000000e+00  6.73920554e-08 
 0.00000000e+00  6.73920554e-08 
 0.00000000e+00  8.80050687e-08 
 0.00000000e+00  8.80050687e-08 
 0.00000000e+00  1.43119352e-07 
 0.00000000e+00  1.43119352e-07 
 0.00000000e+00  1.43119352e-07 `

RTimothyEdwards, 2021-12-19:
Well, for one, generally you don't want to use `mc` with `mc_mm_switch`.  The `mc` option is for process variation.  The `mc_mm_switch` is for device mismatch.  Generally speaking, one doesn't do both in the same simulation.  Mismatch parameters are given at corners, and there are specific sections in the sky130.lib.spice for those.

But apart from that, I'm not sure it's valid to make multiple runs from inside an ngspice control block for monte carlo runs.  I would at least try just making multiple ngspice runs and concatenating the results from all of them outside of ngspice.

frbvianna, 2022-04-04:
@RTimothyEdwards Would you please provide more thoughts into why you think it's not valid to make multiple runs from inside the ngspice control block for the MC runs?
I'm asking because that is how the ngspice manual exemplifies the MC runs - looping inside the control block. Though, I am seeing simulation results that are a bit odd in between iterations.