fossi-foundation/open-pdks #203: open_pdks creates LEF entities with duplicate pins and multiple ports per pin

antonblanchard, 2021-12-22:
I have a recent version of magic:

```
magic --version
8.3.242
```

And a recent checkout of open_pdks (bf1da958f77c). I'm hitting pin access issues with `sky130_fd_sc_hd__fa_2` cells.

The installed `sky130_fd_sc_hd__fa_2` LEF has 2 `PIN A`'s and multiple ports within these pins. Based on our discussion in  https://github.com/The-OpenROAD-Project/OpenROAD/issues/580 it sounds like that is invalid.

If I take the LEF from the skywater PDK, and read/write the LEF in magic, it writes out a LEF without duplicate pins and without mutliple ports per pin. I'm a bit confused how we ended up with this:

```
MACRO sky130_fd_sc_hd__fa_2
  CLASS CORE ;
  FOREIGN sky130_fd_sc_hd__fa_2 ;
  ORIGIN 0.000 0.000 ;
  SIZE 8.280 BY 2.720 ;
  SYMMETRY X Y R90 ;
  SITE unithd ;
  PIN A
    DIRECTION INPUT ;
    USE SIGNAL ;
    ANTENNAGATEAREA 0.631500 ;
    PORT
      LAYER met1 ;
        RECT 1.465 1.260 1.755 1.305 ;
        RECT 2.845 1.260 3.135 1.305 ;
        RECT 4.705 1.260 4.995 1.305 ;
        RECT 6.085 1.260 6.375 1.305 ;
        RECT 1.465 1.120 6.375 1.260 ;
        RECT 1.465 1.075 1.755 1.120 ;
        RECT 2.845 1.075 3.135 1.120 ;
        RECT 4.705 1.075 4.995 1.120 ;
        RECT 6.085 1.075 6.375 1.120 ;
    END
    PORT
      LAYER met1 ;
        RECT 1.925 1.600 2.215 1.645 ;
        RECT 3.785 1.600 4.075 1.645 ;
        RECT 6.545 1.600 6.835 1.645 ;
        RECT 1.925 1.460 6.835 1.600 ;
        RECT 1.925 1.415 2.215 1.460 ;
        RECT 3.785 1.415 4.075 1.460 ;
        RECT 6.545 1.415 6.835 1.460 ;
    END
    PORT
      LAYER met1 ;
        RECT 0.000 -0.240 8.280 0.240 ;
    END
    PORT
      LAYER met1 ;
        RECT 0.000 2.480 8.280 2.960 ;
    END
  END A
  PIN A
    DIRECTION INPUT ;
    USE SIGNAL ;
    ANTENNAGATEAREA 0.157500 ;
    PORT
      LAYER li1 ;
        RECT 1.245 1.275 1.505 1.325 ;
        RECT 1.245 0.995 1.755 1.275 ;
      LAYER mcon ;
        RECT 1.525 1.105 1.695 1.275 ;
    END
    PORT
      LAYER li1 ;
        RECT 2.685 1.030 3.075 1.360 ;
      LAYER mcon ;
        RECT 2.905 1.105 3.075 1.275 ;
    END
    PORT
      LAYER li1 ;
        RECT 4.720 0.955 5.080 1.275 ;
      LAYER mcon ;
        RECT 4.765 1.105 4.935 1.275 ;
    END
    PORT
      LAYER li1 ;
        RECT 6.105 0.995 6.960 1.275 ;
      LAYER mcon ;
        RECT 6.145 1.105 6.315 1.275 ;
    END
  END A
  PIN B
    DIRECTION INPUT ;
    USE SIGNAL ;
    ANTENNAGATEAREA 0.157500 ;
    PORT
      LAYER li1 ;
        RECT 1.645 1.445 2.155 1.690 ;
      LAYER mcon ;
        RECT 1.985 1.445 2.155 1.615 ;
    END
    PORT
      LAYER li1 ;
        RECT 3.655 1.435 4.070 1.745 ;
      LAYER mcon ;
        RECT 3.845 1.445 4.015 1.615 ;
    END
    PORT
      LAYER li1 ;
        RECT 6.150 1.445 6.835 1.735 ;
      LAYER mcon ;
        RECT 6.605 1.445 6.775 1.615 ;
    END
  END B
  PIN CIN
    DIRECTION INPUT ;
    USE SIGNAL ;
    ANTENNAGATEAREA 0.475500 ;
    PORT
      LAYER li1 ;
        RECT 2.325 1.570 3.415 1.740 ;
        RECT 5.670 1.685 5.920 1.955 ;
        RECT 2.325 1.275 2.495 1.570 ;
        RECT 2.125 1.105 2.495 1.275 ;
        RECT 3.245 1.250 3.415 1.570 ;
        RECT 4.295 1.515 5.920 1.685 ;
        RECT 4.295 1.435 4.655 1.515 ;
        RECT 4.295 1.250 4.465 1.435 ;
        RECT 5.670 1.355 5.920 1.515 ;
        RECT 3.245 0.965 4.465 1.250 ;
    END
  END CIN
  PIN VGND
    DIRECTION INOUT ;
    USE GROUND ;
    SHAPE ABUTMENT ;
    PORT
      LAYER li1 ;
        RECT 0.180 0.085 0.350 0.565 ;
        RECT 1.020 0.085 1.350 0.465 ;
        RECT 2.755 0.085 3.085 0.445 ;
        RECT 3.695 0.085 4.025 0.490 ;
        RECT 4.535 0.085 4.865 0.445 ;
        RECT 6.780 0.085 7.110 0.445 ;
        RECT 7.895 0.085 8.065 0.565 ;
        RECT 0.000 -0.085 8.280 0.085 ;
      LAYER mcon ;
        RECT 0.145 -0.085 0.315 0.085 ;
        RECT 0.605 -0.085 0.775 0.085 ;
        RECT 1.065 -0.085 1.235 0.085 ;
        RECT 1.525 -0.085 1.695 0.085 ;
        RECT 1.985 -0.085 2.155 0.085 ;
        RECT 2.445 -0.085 2.615 0.085 ;
        RECT 2.905 -0.085 3.075 0.085 ;
        RECT 3.365 -0.085 3.535 0.085 ;
        RECT 3.825 -0.085 3.995 0.085 ;
        RECT 4.285 -0.085 4.455 0.085 ;
        RECT 4.745 -0.085 4.915 0.085 ;
        RECT 5.205 -0.085 5.375 0.085 ;
        RECT 5.665 -0.085 5.835 0.085 ;
        RECT 6.125 -0.085 6.295 0.085 ;
        RECT 6.585 -0.085 6.755 0.085 ;
        RECT 7.045 -0.085 7.215 0.085 ;
        RECT 7.505 -0.085 7.675 0.085 ;
        RECT 7.965 -0.085 8.135 0.085 ;
    END
  END VGND
  PIN VNB
    DIRECTION INOUT ;
    USE GROUND ;
    PORT
      LAYER pwell ;
        RECT 0.010 0.785 1.360 1.015 ;
        RECT 6.765 0.785 8.235 1.015 ;
        RECT 0.010 0.105 8.235 0.785 ;
        RECT 0.145 -0.085 0.315 0.105 ;
    END
  END VNB
  PIN VPB
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER nwell ;
        RECT -0.190 1.305 8.470 2.910 ;
    END
  END VPB
  PIN VPWR
    DIRECTION INOUT ;
    USE POWER ;
    SHAPE ABUTMENT ;
    PORT
      LAYER li1 ;
        RECT 0.000 2.635 8.280 2.805 ;
        RECT 0.180 1.795 0.350 2.635 ;
        RECT 0.980 2.275 1.310 2.635 ;
        RECT 2.755 2.275 3.085 2.635 ;
        RECT 3.695 1.915 4.025 2.635 ;
        RECT 4.535 2.275 4.865 2.635 ;
        RECT 6.890 2.275 7.220 2.635 ;
        RECT 7.815 1.795 7.985 2.635 ;
      LAYER mcon ;
        RECT 0.145 2.635 0.315 2.805 ;
        RECT 0.605 2.635 0.775 2.805 ;
        RECT 1.065 2.635 1.235 2.805 ;
        RECT 1.525 2.635 1.695 2.805 ;
        RECT 1.985 2.635 2.155 2.805 ;
        RECT 2.445 2.635 2.615 2.805 ;
        RECT 2.905 2.635 3.075 2.805 ;
        RECT 3.365 2.635 3.535 2.805 ;
        RECT 3.825 2.635 3.995 2.805 ;
        RECT 4.285 2.635 4.455 2.805 ;
        RECT 4.745 2.635 4.915 2.805 ;
        RECT 5.205 2.635 5.375 2.805 ;
        RECT 5.665 2.635 5.835 2.805 ;
        RECT 6.125 2.635 6.295 2.805 ;
        RECT 6.585 2.635 6.755 2.805 ;
        RECT 7.045 2.635 7.215 2.805 ;
        RECT 7.505 2.635 7.675 2.805 ;
        RECT 7.965 2.635 8.135 2.805 ;
    END
  END VPWR
  PIN COUT
    DIRECTION OUTPUT ;
    USE SIGNAL ;
    ANTENNADIFFAREA 0.445500 ;
    PORT
      LAYER li1 ;
        RECT 0.600 1.950 0.810 2.465 ;
        RECT 0.565 1.780 0.810 1.950 ;
        RECT 0.565 1.585 0.735 1.780 ;
        RECT 0.085 1.415 0.735 1.585 ;
        RECT 0.085 0.905 0.370 1.415 ;
        RECT 0.085 0.735 0.690 0.905 ;
        RECT 0.520 0.485 0.690 0.735 ;
        RECT 0.520 0.315 0.850 0.485 ;
    END
  END COUT
  PIN SUM
    DIRECTION OUTPUT ;
    USE SIGNAL ;
    ANTENNADIFFAREA 0.523500 ;
    PORT
      LAYER li1 ;
        RECT 7.395 1.965 7.565 2.465 ;
        RECT 7.395 1.795 7.645 1.965 ;
        RECT 7.475 1.585 7.645 1.795 ;
        RECT 7.475 1.415 8.195 1.585 ;
        RECT 7.970 0.905 8.195 1.415 ;
        RECT 7.475 0.735 8.195 0.905 ;
        RECT 7.475 0.485 7.725 0.735 ;
        RECT 7.395 0.255 7.725 0.485 ;
    END
  END SUM
  OBS
      LAYER li1 ;
        RECT 1.900 2.080 2.245 2.465 ;
        RECT 1.305 1.910 2.245 2.080 ;
        RECT 2.415 2.105 2.585 2.465 ;
        RECT 3.255 2.105 3.490 2.465 ;
        RECT 2.415 1.935 3.490 2.105 ;
        RECT 4.195 2.105 4.365 2.465 ;
        RECT 5.035 2.105 5.205 2.465 ;
        RECT 5.485 2.125 6.685 2.465 ;
        RECT 4.195 1.935 5.205 2.105 ;
        RECT 6.515 2.105 6.685 2.125 ;
        RECT 6.515 1.935 7.180 2.105 ;
        RECT 1.305 1.665 1.475 1.910 ;
        RECT 0.940 1.600 1.475 1.665 ;
        RECT 0.905 1.495 1.475 1.600 ;
        RECT 7.010 1.640 7.180 1.935 ;
        RECT 0.905 1.430 1.110 1.495 ;
        RECT 7.010 1.470 7.300 1.640 ;
        RECT 0.905 1.245 1.075 1.430 ;
        RECT 0.540 1.075 1.075 1.245 ;
        RECT 7.130 1.245 7.300 1.470 ;
        RECT 0.905 0.825 1.075 1.075 ;
        RECT 5.250 0.955 5.935 1.125 ;
        RECT 1.925 0.825 2.165 0.935 ;
        RECT 0.905 0.655 2.165 0.825 ;
        RECT 1.535 0.255 2.165 0.655 ;
        RECT 2.415 0.615 3.425 0.785 ;
        RECT 2.415 0.255 2.585 0.615 ;
        RECT 3.255 0.255 3.425 0.615 ;
        RECT 4.195 0.615 5.205 0.785 ;
        RECT 5.420 0.765 5.935 0.955 ;
        RECT 7.130 1.075 7.800 1.245 ;
        RECT 7.130 0.825 7.300 1.075 ;
        RECT 4.195 0.255 4.365 0.615 ;
        RECT 5.035 0.255 5.205 0.615 ;
        RECT 6.380 0.655 7.300 0.825 ;
        RECT 5.540 0.505 5.710 0.595 ;
        RECT 6.380 0.505 6.550 0.655 ;
        RECT 5.540 0.255 6.550 0.505 ;
      LAYER mcon ;
        RECT 1.985 0.765 2.155 0.935 ;
        RECT 5.685 0.765 5.855 0.935 ;
      LAYER met1 ;
        RECT 1.925 0.920 2.215 0.965 ;
        RECT 5.625 0.920 5.915 0.965 ;
        RECT 1.925 0.780 5.915 0.920 ;
        RECT 1.925 0.735 2.215 0.780 ;
        RECT 5.625 0.735 5.915 0.780 ;
  END
END sky130_fd_sc_hd__fa_2
```

antonblanchard, 2021-12-22:
I think it's a bug in magic, bad things start happening from this commit:

```
commit 43bb499bcf0a2e18114eb9187783c63ddf4e93d1
Author: Tim Edwards <tim@opencircuitdesign.com>
Date:   Sun Dec 12 22:09:31 2021 -0500

    Corrected an issue with the limited bitfield dedicated to port numbers;
    this limited ports to 16384, which seemed reasonable at the time.
    However, the sky130_sram_macro layouts connect power and ground in a
    way that when coupled with "extract unique" can generate tens of
    thousands of ports and overrun the bit field, showing that automation
    can do the unexpected.  The solution was to split out the port number
    from the label record as its own 32-bit value.
```

RTimothyEdwards, 2021-12-22:
It shouldn't have anything to do with that commit, but I'll check what's going on.

RTimothyEdwards, 2021-12-22:
Okay, I sort of get it.  Must have missed something in the "lef write" routine. . .

RTimothyEdwards, 2021-12-22:
Thanks for the report.  I tracked down the issue to "lef read" and fixed it.  The corrected version is 8.3.243 (commit 716848067eefb7a54aa0719e1c2c54e7cd2bda49).

antonblanchard, 2021-12-22:
Thanks @RTimothyEdwards that fixed it.

antonblanchard, 2022-01-20:
Since this was verified to be fixed, I'll close it out.

closed 2022-01-20 by antonblanchard