fossi-foundation/open-pdks #204: bad links to magicrc

d-m-bailey, 2021-12-27:
I don't know if this is critical or not, but I notice some bad links to `.magicrc` in the pdk.
Executing the following in the openlane installation directory
```
ls -l `find pdks/sky130A -name .magicrc
```
gives me a bunch of lines like this
```
/home/user/OpenLane/pdks/sky130A/libs.ref/sky130_sram_macros/maglef/.magicrc -> /home/user/OpenLane/pdks/open_pdks/sky130/sky130A/libs.tech/magic/sky130A.magicrc 
```
Looks like there's an extra `sky130A` in the link.

I see this in the older
```
commit c5730b574461889c82858b08d12ba42423d9c2cb (HEAD)
Author: Tim Edwards <tim@opencircuitdesign.com>
Date:   Wed Nov 10 16:26:06 2021 -0500
```
and the newer
```
commit 476f7428f7f686de51a5164c702629a9b9f2da46 (HEAD, tag: 1.0.255)
Author: Tim Edwards <tim@opencircuitdesign.com>
Date:   Thu Dec 16 19:59:35 2021 -0500
```

Apologies if this is not a concern or has been fixed.

RTimothyEdwards, 2022-01-05:
Fixed today. Sorry about dropping the ball on that one.

RTimothyEdwards, 2022-01-05:
The specific issue is that the install script `staging_install.py` claims to remove the temporary `.magicrc` files that were used for the build, but it wasn't removing them.  The symbolic link is pointing to the staging directory, not the target directory.  Openlane has the target directory rooted in the same place as open_pdks, so the staging directory and target directory happen to differ by only one path component.  The problem would be much more obvious if it were installing to the usual /usr or /usr/local destination.

olofk, 2022-01-25:
Can this be closed now?

RTimothyEdwards, 2022-01-25:
Yes, I'll close it now.

closed 2022-01-25 by RTimothyEdwards