fossi-foundation/open-pdks #206: Extract of SC layout from .mag needs MAGIC_EXT_USE_GDS=1

hpretl, 2021-12-28:
When I extract the netlist from a `.mag` layout block which has been generated in SKY130 using `OpenLane`, and I use `netgen` to compare it to the powered Verilog netlist, then it fails by default due to the filler cells.

When I set `export MAGIC_EXT_USE_GDS=1`, then it works and the LVS passes.

Question: Why is this not the default behavior, and maybe the switch should be renamed to reflect the issue that not only an extract in `magic` after GDS read-in but also when starting from the `.mag` file needs this switch?

The relevant code lives in `setup.tcl` in `$PDK_ROOT/sky130A/libs.tech/netgen`:

```
if { [info exist ::env(MAGIC_EXT_USE_GDS)] && $::env(MAGIC_EXT_USE_GDS) } {
    foreach cell $cells1 {
        if {[regexp {sky130_fd_sc_[^_]+__fill_[[:digit:]]+} $cell match]} {
            ignore class "-circuit1 $cell"
        }
        if {[regexp {sky130_fd_sc_[^_]+__tapvpwrvgnd_[[:digit:]]+} $cell match]} {
            ignore class "-circuit1 $cell"
        }
        if {[regexp {sky130_ef_sc_[^_]+__fakediode_[[:digit:]]+} $cell match]} {
            ignore class "-circuit1 $cell"
        }
    }
```

For the extract in `magic` I use the procedure below, perhaps in an SC layout I should not extract the standard cells? However, in this way I can also extract a mixed analog/digital design.

```
extract all
ext2spice lvs
ext2spice -o <file.spice>
```

hpretl, 2021-12-28:
BTW, if I try to use the `ext2spice subcircuit descend off` in the extract procedure above, there is no difference, and I still have the subcircuits of the standard cells in the SPICE netlist.

d-m-bailey, 2021-12-28:
@hpretl Originally, the fill, tap, and fakediode cells were ignored unconditionally. LVS had two modes - maglef based (no devices) and GDS based. With the maglef based netlist, it is possible to (and arguably should) compare the connections of the fill, tap, and fakediode cells because both the layout and the netlist are black boxed. With the GDS based (or as you indicated the mag based) version, the fill, tap, and fakediode cells were extracted as instances of non-existent subcircuits which caused LVS mismatches. At least, that's my understanding of how things got the way they were. 

So netgen needs to know whether it's running device level LVS or block level LVS. I agree with your suggestion of renaming the variable. I'm in the middle of proposing some changes to these scripts. Any suggestions?
Maybe `LVS_DEVICE_LEVEL` or `LVS_BLOCK_LEVEL`?

hpretl, 2021-12-28:
How about LVS_ENABLE_MAGLEF? The default should be for GDS and MAG extracts (with this variable unset), and when this variable is set the LVS on the MAGLEF with the bleach boxes works.