fossi-foundation/open-pdks #209: Magic extraction fails due to li1 pin and li1 obstruction overlaps

antonblanchard, 2022-01-21:
I'm seeing many warnings when doing LEF/DEF level extraction:

```
feedback add "Illegal overlap between obsli1 and locali (types do not connect)" medium
```

Many of them are between `sky130_fd_sc_hd__buf_2` and `sky130_fd_sc_hd__decap_12.lef`. Looking at  `sky130_fd_sc_hd__decap_12.lef`, the li1 layer is made up entirely of an obstruction not a pin, which explains why we are seeing the errors. The original lef in the upstream PDK shows the li1 layer as a pin.

antonblanchard, 2022-01-21:
I also notice that we are using decap cells for almost all our fill. Just wanted to confirm that that is expected.

RTimothyEdwards, 2022-01-22:
I'm not sure when this happened, or why.  The code that magic runs to generate the LEF views should be consistent in its policy (the original SkyWater cells are very inconsistent) in either putting both the local interconnect and metal1 on the power rails into the pin list, or just the metal1.  For reasons I can't ascertain (yet), the buf_1 and buf_2 cells (possibly others) are including local interconnect in the power rails pin lists.  I'll investigate.

RTimothyEdwards, 2022-01-22:
If you use all decap, make sure you use `sky130_ef_sc_hd__decap_12` instead of `sky130_fd_sd_hd__decap_12`.  The SkyWater decap cells have too much local interconnect, and heavy use of fill will violate the local interconnect pattern density rule.  My version of the decap_12 cell pares back the local interconnect to be about 50% of the cell, which works better for satisfying the pattern density rules.  The other choice is to mix about half and half of fill and decap cells.

RTimothyEdwards, 2022-01-24:
@antonblanchard :  I have tracked this down to a minor difference in labeling between seven of the cells in the HD library (including buf_1 and buf_2) and all the others, plus an error in magic when dealing with labels during "lef write" when using the options applied by the open_pdks installation for the SkyWater standard cell libraries.  I fixed the error in magic.  Therefore if magic gets updated to version `8.3.262` and open_pdks installation is re-run, then the HD library abstract views should be correct, and the problem will no longer occur.

antonblanchard, 2022-01-25:
@RTimothyEdwards Thanks! I confirmed that this does fix the issue for me.

closed 2022-01-25 by antonblanchard